Ultimaker/Cura #7: Do not perform a slice if there are error values in the settings

awhiemstra, 2015-05-15:
Platform: Arch Linux 64-bit
Qt: 5.4.1
PyQt: 5.4.1
Diplay Driver: xf86-video-intel 2.99.917

Steps to Reproduce:
* Load a model.
* Switch to "Advanced" mode.
* Open the "Quality" tab
* Set "Layer Height" to 0

Actual Results:
* The text box turns red to indicate an error but then slicing starts and the engine will crash every time until the Layer Height is set back to something > 0.

Expected results:
* The text box turns red and slicing does not start.

Additional Information:
* After some discussion with BagelOrb we decided that we should assume error values are incorrect values and not slice if there are any, whereas warning values are things we think are invalid values but which may work. We should make sure we have correct error and warning values on settings.

nallath, 2015-05-15:
This is a big change compared to original cura, where red values are assumed to be wrong, but still sent to the engine. It won't 'block'.

awhiemstra, 2015-05-15:
We tested this with layer height in the legacy Cura. With a layer height of 0 legacy Cura will not slice. In addition, right now the difference between the two things is rather vague, so this gives the actual values proper definition.

commit 1471a681 (2015-05-27): Do not slice if there is any setting with an error value

Fixes #7
closed 2015-06-03 by awhiemstra