Ultimaker/Cura #9: Wire Printing in 15.06

flouSH, 2015-05-16:
Hi @BagelOrb ,

can you quickly comment on the Wire Printing feature in 15.06? We tried to get it working, enabled it - but didn't see any changes in the gcode (in Cura visualisation didn't work (via Parallels on a Mac <-- that might be the problem?!)), but also in Repetier Host it doesn't show any wire printing. Just normal outlines.

Any trick or is it not ready for alpha testing yet?

Thanks!
Florian

BagelOrb, 2015-05-18:
It should be ready for alpha testing. The visualisation can't handle the gcode, however. I will look into why it is not working.

BagelOrb, 2015-05-18:
Problem is there are two settings files now: Cura/resources/settings/fdmprinter.json and CuraEngine/fdmprinter.json
For the engine the setting name has changed.
However, the settings file for Cura is frozen for the moment.

You can test wire printing via the command line; this is what I run:
./build/CuraEngine -v -j fdmprinter.json -s mesh_position_x=115.0 -s mesh_position_y=112.5 -s wireframe_enabled=1 -o "output/test.gcode" "Cura_test.stl"; ed -s output/test.gcode <<< $'1i\n;FLAVOR:UltiGCode\n;TIME:10000000\n;MATERIAL:2000\n;MATERIAL2:0\n.\nwq'

However, currently wirePrinting suffers from another issue: https://github.com/Ultimaker/CuraEngine/issues/202

awhiemstra, 2015-05-27:
This should be fixed with recent builds of 15.06. There was a (pretty severe) bug in the engine that caused all boolean settings to always evaluate to false. The issue in the engine was fixed and wire printing should now work too.

closed 2015-05-27 by awhiemstra