Ultimaker/Cura #11: I can't change layer height

MiguelAngelLV, 2015-05-19:
The layer height is always of 0.095. When I try change I can't type a dot and I write a comma the field is become in red.

If I only delete 095 and wite 2 (0.2) not change anything, the gcode is for 0.095. If I change the .cfg file, set 0.2 for height_layer and open Cura, the value is change to 0.095.

I use the code from today and branchs 15.06

MiguelAngelLV, 2015-05-19:
I've noticed that when I change the quality "simple mode" the height of layer changes. It seems that not overwrite option

MiguelAngelLV, 2015-06-25:
The bug is still in last version. The values of profiles override the manual value.

awhiemstra, 2015-06-25:
Oh, seems I misread the original issue, since I thought it was about the dot-versus-comma we have has issues with before.

In regards to simple mode overwriting the set layer value, I will need some more comments from people since I have heard from other people that they actually expect that to happen.

MiguelAngelLV, 2015-06-25:
Um... a checkbox to "force" a user manual configuration?

awhiemstra, 2015-06-25:
So there was a bug that I fixed where the simple mode settings would override the set layer height even if we open in Advanced mode. This is now fixed, though values will still be overridden when switching to simple mode and back. That however, will probably not be fixed any time soon.

MiguelAngelLV, 2015-06-25:
So, If I use advance mode the layer height will not overrride, not? Ok for
me

awhiemstra, 2015-06-25:
Yeah, if you switch to advanced and do not switch back, it will remember the layer height properly.

nallath, 2015-07-08:
@MiguelAngelLV Is this still an issue in the latest version?

MiguelAngelLV, 2015-07-08:
@nallath the issue is solved, thanks.

closed 2015-07-08 by MiguelAngelLV