Ultimaker/Cura #12: Add   "disable auto slicer" option

MiguelAngelLV, 2015-05-23:
When I try order big models in the bed, each time I move a model Cura slice all model and "block" the app.

Could you add a "Disable auto slicer" option and add a button to slicer when you finish order the models?

Dim3nsioneer, 2015-05-23:
@MiguelAngelLV This request has been made for many, many times. Ultimaker R&D has clearly decided to keep the autoslicing.
I guess this will stay true with the new Cura which we are actually dealing with here (see Daid/Cura for versions up to 15.02.1. There you can move one of your objects slightly out of the building platform so it becomes grey. Autoslicing then stops.

closed 2015-05-24 by daid

nallath, 2015-05-27:
It should be possible to write a plugin that does this. We have a contest for new plugins, you can find it on https://cura-ultimaker.com/