Ultimaker/Cura #13: Plugin develop guide

MiguelAngelLV, 2015-05-27:
We need a "guide" to develop plugins to new systems. I [have a plugin][1] I would like port to new system but I don't know how start

[1]: https://github.com/MiguelAngelLV/marlin-lcd-estimate-time

Dim3nsioneer, 2015-05-28:
+1
I'm in contact with @awhiemstra about future 'plugins' and having any easy API for postprocessing scripts again.

quillford, 2015-05-28:
Fortunately, it seems like EVERYTHING is a plugin - even the USB communications.

MiguelAngelLV, 2015-05-28:
And the "concept" of old plugins? The plugins that only modify the gcode.
We would need I few examples to port the current plugins.

awhiemstra, 2015-05-28:
The current idea is to develop a post-processing plugin that implements similar functionality as the old post-processing plugin structure. I have been discussing the details with @Dim3nsioneer in private, but we can also move that discussion here for more input.

MiguelAngelLV, 2015-05-28:
Perfect

Dim3nsioneer, 2015-05-28:
I would appreciate to have other opinions than just mine... ;-)

quillford, 2015-05-28:
@Dim3nsioneer well, that should also increase the average quality of plugins, but if a plugin is poorly written, you always have the option of not using it

quillford, 2015-05-28:
@Dim3nsioneer you also have to keep in mind that many (if not most) users of repraps and Cura are programmers :P

Dim3nsioneer, 2015-05-28:
@quillford User segments have significantly changed since autumn 2014...

daid, 2015-05-29:
@quillford RepRaps is only tiny segment of the Cura users. Programmers are only a minuscule faction of the Cura users.

nallath, 2015-06-03:
I've started with this. You can find the first bits here https://github.com/Ultimaker/Uranium/wiki/Plugin-Types

nallath, 2015-06-04:
Some more info;
https://github.com/Ultimaker/Uranium/wiki/Creating-plugins
https://github.com/Ultimaker/Uranium/wiki/Backend-&-Plugin-structure

The documentation will also be updated soon. Please let me know if this is what you guys need to get started.

MiguelAngelLV, 2015-06-04:
@nallath I don't know yet how I start to develop a post-processing gcode plugin...

nallath, 2015-06-05:
The post-processing plugin type is not yet implemented, so there is no documentation to write in that regard, sorry.

The hackish solution would be to to use an extension plugin. I will discuss this with @awhiemstra. Will get back to you.

awhiemstra, 2015-06-05:
So these are the preliminary requirements we came up with:

- Post processing files should be a single python file with a single entry point.
- Post processing files should work on a list of strings representing GCode, not on files as they used to do.
- Post processing files will not be sandboxed or otherwise restricted in their execution until there is clear proof that people are exploiting things this way.
- There should be a way to configure post processing files from the post processing plugin.
- It should be possible to run multiple post processing files after each other, including multiple instances of the same plugin.

gearsawe, 2015-07-08:
I have had some success but it is all a hack really and the only way I have found to make this work for what I need, those who own a M3D Mirco and want to use 15.06
I modified the Original Gcode writer which access settings from those set thru the printer config. I know the printer config is not where these setting should be but in this case yes. Since the settings are for making changes to gcode which are specific to the printer.
I tried to make a separate GcodeWriterM3D along side of the GCodeWrite but all it seem to do is use the original and ignore mine.
@BagelOrb thanks for the example plugin. This give us something to dwell on. And still is not easy to figure this all out.

What Would really be nice if:
-Pose Process Plugs are run at save.
-You could make your own setting there.
-Setting could be saved to an ini file which could be loaded later at the time the plugin is run. (This of course to be done thru normal python script. And up to the programmer to make their own saving format.)
-Not only would being able to run multiple posting processing files but ability to chose the order.

quillford, 2015-07-08:
@gearsawe I heard that M3D's software uses CuraEngine. Why does it need a post processor?

gearsawe, 2015-07-08:
@quillford Well their spooler does post processing of the gcode which is 100% not correct. They do some really weird stuff to the gcode before it is sent to the printer and I found a way "fix" things. I proccess the file before they do so their program do not mess things up. M3D print does accept gcode. They have their own software which basically uses the old Cura to create the gcode based on a bunch of preset. It would seem they refuse to fix their post processing since they fail to see the problem. I hope in time to completely bypass their software and use Cura to communicate to the printer making it much better printer. I am just doing baby steps.

Dim3nsioneer, 2015-07-16:
@awhiemstra @nallath Can you comment on the status of a post processing script system? Do you request action from the community or will you come up with some code?

awhiemstra, 2015-07-17:
Now that 15.06 is out we are once more doing work on feature development. I have started working on an overhaul of the output device API as that is also sorely needed. After that I plan to create a post processing plugin, which is hopefully going to be somewhere next week. Before that, I would like to know any additional requirements we did not yet think of.

gearsawe, 2015-07-20:
@awhienstra I would also guess the post processing should also have a way for it save and recall settings. I hacked a plugin together for my own personal use and stored settings in the printer config. The problem is I wanted the same plugin settings to be used across multiple printer configs. This make it kind of difficult with my setup. Secondly if I disable my plugin then close Cura then Open re-enable the plugin again I have lost my setting and the are back to the defaults. I am sure with some time I could make something stores setting in a INI file then save and recall from there.  My goal it to be able to have multiple profiles which can be chosen from within the same plugin. Hope this make sense. Still very new to Cura but love it so far.

nallath, 2015-07-21:
You could use the preferences for this. Take a look at my printer cost calculation plugin.

nallath, 2015-07-27:
I've started work on this. See https://github.com/nallath/PostProcessingPlugin

nallath, 2015-07-29:
Just a quick update; A first sort of working version is up. Check it out.

nickthetait, 2015-07-29:
Good to hear. Does it do something specific or is it a generic demo of how plugins can be implemented?

nallath, 2015-07-30:
Its plugin that runs scripts (which sorta are plugins on its own). I've added the pause at height script that used to be in the old Cura, so it actually does something.

closed 2015-07-31 by nallath