Ultimaker/Cura #15: [OS X] 3D view only uses part of available space

ErikDeBruijn, 2015-05-28:
As a user I'd like to see my platform and 3D models in all of their glory :)
The 3D view only uses a part of the space that is available.

Where: Main interface

Actions: Using [the latest](http://software.ultimaker.com/Cura_closed_beta/15.06/Cura-15.05.92-Darwin.dmg) release for Darwin.

Behavior: Just upon opening. When resizing the window there are changes, but it always cuts off the view which uses a part of the available screen real estate.

Expected behavior: The window should have  the 3D view using the max of the size available.

Screenshot:
[image]

Specs: OS X 10.10.3

ErikDeBruijn, 2015-06-03:
It's present on this Macbook with Retina, indeed. I saw it worked fine on an older Mac without retina.

I can test for you if you have a release that should work with Retina screens.

awhiemstra, 2015-06-03:
This should be fixed with the new 15.05.93 build

oliv3r, 2015-06-04:
So this might be a hidpi issue. Ill check under linux with hidpi to see of it happend there aswell

ErikDeBruijn, 2015-06-05:
@awhiemstra the .93 version fixes it! Great job!

closed 2015-06-05 by ErikDeBruijn