Ultimaker/Cura #16: Warn if estimated filament used is too much

BagelOrb, 2015-05-28:
Feature request:
Warn if estimated filament used exceeds standard amount of filament on a roll

Dim3nsioneer, 2015-05-28:
Which is how much? ;-) Faberdashery rolls are e.g. longer than Colorfabb spools...

quillford, 2015-05-28:
Perhaps the warning shouldn't appear unless the user set it (which could be part of the initial setup process) because some companies use mass and others use length.

daid, 2015-05-29:
Normal 2.85mm rolls of 750g contain about 80-100 meters of material.

BagelOrb, 2015-05-31:
Normal: use defaults as described by daid
Enhancement: make the threshold depend on the material profile

Fass-ohne-Hoden, 2015-07-13:
For PLA, I usually buy spools with 1 kg of material, but I've also had other stuff on spools with half a Kilo and 750 g. Some sell 450 g spools, I think. There are several common sizes. And then, density varies quite a bit within the common range of materials. Moreover, pretty much all users will sooner or later switch between spools of different sizes when switching materials.

So, if there was such a function, it should neither be set in stone nor be designed into the program as set-once-and-never-again thing. Nice would be to have the length of material also shown as volume (as suggested in another issue) AND as mass for different materials that you could either tick or untick in the settings bar (and, preferably, also manage/edit/add/delete yourself in the options dialogue)

fieldOfView, 2017-10-01:
Cura: Cold Case dept.

ianpaschal, 2018-05-22:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-22 by ianpaschal