Ultimaker/Cura #18: Adding new storage_device plugin

ad1217, 2015-06-01:
I was trying to write a write a device that implemented StorageDevice (for uploading to Octoprint), but it seems that everything is hard-coded to use LocalFileStorage. Is this temporary? Or should I be doing this some other way?

nallath, 2015-06-03:
Storage device is indeed the way to go for uploading to OctoPrint (or atleast, you understood correctly with regards that is was intended to be used like that ;)). Due to time constraints on our end, we did a bit of a nasty by hardcoding it. 
We definitely should be handling this better.

awhiemstra, 2015-06-03:
The entire storage device API is rather broken at the moment unfortunately. Originally, the idea behind it was that you can implement different types of devices that can then handle the specifics of file uploads etc. However, during development it turned out that the way this is currently implemented is pretty broken. For starters, it should be the storage device that decides what file formats to support, whereas this is currently reversed. Additionally, there should be plugins that provide a list of files that can be loaded as well as plugins that can write files somewhere. 

To have something that at least mostly works for Cura, I implemented the output_device related functions in CuraApplication.py . These are currently used by the Save button to determine where to save the file. It has a reasonably limited API that eventually I would like to extend so it can be the generic API for output plugins. This is probably what you want to use for the OctoPrint plugin as well, this is also what the USB printing plugin uses at the moment.

Of course, it would be even better to revamp the entire storage_device API, but that takes quite a bit of time and effort. I have some plans for this, but I will not have time to do it soon I'm afraid.

closed 2015-06-04 by nallath