Ultimaker/Cura #19: [OS X] Can't rotate or move objects

ErikDeBruijn, 2015-06-01:
I'm trying to move objects that I've just opened and place them somewhere different on the platform. With rotating, it's also not clear how to rotate it.

Where: Main interface

Actions: Using the 15.05.92 Darwin release.

Behavior:
- I click and hold the trackpad to drag the object to it's new place. Also tried "selecting" it first, but the object doesn't move. Also, when clicking on the rotate icon I don't see anything change (except the icon gets highlighted and more icons appear), don't see any controls.
 - Scale to max also doesn't do anything.
 - Mirror, ditto.

Expected behavior: 
 - [ ] Dragging objects should be possible at all times.
 - [ ] When clicking on rotate I want to see controls in the 3D view and the ability to use them to change the orientation of a part.
 - [ ] Mirror should also help me see how to mirror, and when it mirrors on which axis. Or I'd like to be able to use scale for that. Better leave out the button or implement it.
Screenshot:
[image]

Specs: OS X 10.10.3

Question, I haven't tried the windows version but could these issues be specific to OS X?

awhiemstra, 2015-06-02:
> Question, I haven't tried the windows version but could these issues be specific to OS X?

They are. I also have not seen them on my testing machine. I think this one and #15 are related and your 3D view seems not to update properly. Can you run the application from a terminal and give me the log output?

ErikDeBruijn, 2015-06-03:
This is what I get when running it:

```
➜  cura_build   /Applications/Cura\ Beta/Cura.app/Contents/MacOS/Cura
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,022 - INFO - Loaded plugin ConsoleLogger
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,069 - INFO - Loaded plugin FileLogger
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,323 - INFO - Loaded plugin LocalFileStorage
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,331 - INFO - Loaded plugin MeshView
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,333 - INFO - Loaded plugin WireframeView
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,336 - INFO - Loaded plugin LayerView
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,338 - INFO - Loaded plugin OBJReader
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,339 - INFO - Loaded plugin STLReader
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,341 - INFO - Loaded plugin OBJWriter
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,342 - INFO - Loaded plugin STLWriter
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,345 - INFO - Loaded plugin GCodeWriter
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,348 - INFO - Loaded plugin CameraTool
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,352 - INFO - Loaded plugin MirrorTool
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,364 - INFO - Loaded plugin RotateTool
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,368 - INFO - Loaded plugin ScaleTool
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,370 - INFO - Loaded plugin SelectionTool
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,374 - INFO - Loaded plugin TranslateTool
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,377 - INFO - Checking for new version of cura
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,380 - INFO - Loaded plugin UpdateChecker
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,388 - INFO - Loaded plugin USBPrinting
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,395 - INFO - Could not establish connection on /dev/cu.Bluetooth-Modem: 'Failed to open serial port'. Device is not arduino based.
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,396 - INFO - Could not open port /dev/cu.Bluetooth-Modem
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,399 - INFO - Loaded plugin CuraEngineBackend
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,450 - INFO - Found a new version of the software. Spawning message
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,467 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,467 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,467 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
Jun  3 11:00:45 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:45,468 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
Jun  3 11:00:46 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:46,244 - DEBUG - Loaded theme /Applications/Cura Beta/Cura.app/Contents/Resources/cura/resources/themes/cura
Jun  3 11:00:46 Laptop-Erik.local Cura[23397] <Notice>: file:///Applications/Cura Beta/Cura.app/Contents/Resources/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
Jun  3 11:00:46 Laptop-Erik.local Cura[23397] <Notice>: file:///Applications/Cura Beta/Cura.app/Contents/Resources/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
Jun  3 11:00:46 Laptop-Erik.local Cura[23397] <Notice>: file:///Applications/Cura Beta/Cura.app/Contents/Resources/cura/resources/qml/SidebarSimple.qml:116:22: Unable to assign [undefined] to bool
Jun  3 11:00:46 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:46,824 - INFO - Started engine process: /Applications/Cura Beta/Cura.app/Contents/MacOS/CuraEngine
Jun  3 11:00:46 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:46,945 - DEBUG - Backend connected on port 49674
Jun  3 11:00:47 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:47,108 - INFO - Could not establish connection on /dev/cu.Bluetooth-Incoming-Port: 'Timeout'. Device is not arduino based.
Jun  3 11:00:51 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:51,652 - INFO - Could not establish connection on /dev/cu.PebbleB2D5-SerialPortSe: 'Failed to open serial port'. Device is not arduino based.
Jun  3 11:00:57 Laptop-Erik.local Cura[23397] <Notice>: 2015-06-03 11:00:57,910 - INFO - Could not open port /dev/cu.PebbleB2D5-SerialPortSe
```

The Mac issues are really blocking normal usage (this view issue isn't as critical as the ability to rotate parts and such). 

If it's not specific to my machine it might be quicker to ask any colleague with an Macbook that is in the same building as you are.

awhiemstra, 2015-06-03:
Just checked with a colleagues' MacBook where everything seems to work as intended.

Do you happen to run on a MacBook with high DPI screen (Retina)? That might at least explain the view size issue.

ErikDeBruijn, 2015-06-03:
I've been able to try it in a different Mac and the fact that I thought this was a major bug is because of a combination of factors:
 - The mouse press actually needed to be where the object was supposed to be, not more towards the lower left quarter of the screen.
 - I had not expected that I couldn't drag an object without first clicking on it to select it. I have no idea why you wouldn't want dragging to be possible before selecting the object.

This made it nearly impossible for me to operate on objects.

awhiemstra, 2015-06-03:
The first bit should be fixed now that the viewport sizing is fixed as well. The second bit should also be fixed in the 15.05.93 build though it feels slightly wonky with the camera animation.

ErikDeBruijn, 2015-06-05:
@awhiemstra indeed it is fixed. It's a bit wonky perhaps, but very usable. Scale and rotate also work!

closed 2015-06-05 by ErikDeBruijn

JosepEscobar, 2020-12-10:
Can't rotate again :(

Ghostkeeper, 2020-12-18:
@JosepEscobar Please see https://github.com/Ultimaker/Cura/issues/8745.