Ultimaker/Cura #23: Model modifying tools not scaling

quillford, 2015-06-02:
The tools overlayed on the model for repositioning, rotating, etc. doesn't scale with the rest of the platform.
[image]
[image]

nallath, 2015-06-03:
I believe this is by design, but that does not mean it cannot be changed / improved.

What behaviour would you expect?

awhiemstra, 2015-06-03:
This is indeed intentional and different from legacy Cura. This is behaviour from 3D modelling programs that we copied.

quillford, 2015-06-03:
Ah. It is different from the current Cura release, so I thought it may have been a bug.

closed 2015-06-03 by quillford