Ultimaker/Cura #25: What's the relationship between this cura and https://github.com/daid/Cura

derekhe, 2015-06-04:
I'm confused there are two curas. Don't know what's the difference between them.

In this issue, I want to test the new xy compensation feature, and it pointed out use this repo.
https://github.com/daid/Cura/issues/1194

daid, 2015-06-04:
This is the new, shiny, unreleased frontend for Cura. daid/Cura is the old legacy Cura that everyone knows and loves/hates.

We re-worked the whole GUI code at Ultimaker, because my old code started to become an unmaintainable ball of poo.

derekhe, 2015-06-04:
Cool, I saw right now the code seems compile under cygwin. Is there any instruction to make guys know how to build the entire project? so we can contribute.

nallath, 2015-06-04:
Cura uses the Uranium framework in order to work, which can be found here;
https://github.com/Ultimaker/Uranium 
Why would you run this with cygwin? It should run on mac / linux / windows natively.

derekhe, 2015-06-04:
Thank you @nallath 
I look at the build.sh and so I guess this is for cygwin.
I think I should run cura_app.py to run it

awhiemstra, 2015-06-04:
The build.sh is meant for building a release installer, it is not needed for running the application. As @nallath mentioned, Uranium is needed as dependency. You will also need a recent version of Cura Engine.

closed 2015-06-04 by nallath

arekm, 2015-06-06:
Maybe add explanation sentence to project README.md, so other won't be confused? (as I was, too but fortunately found this issue).