Ultimaker/Cura #26: Decimal point deleted when editing values

Xisle, 2015-06-04:
When editing values in Cura 15.05.93 on OS X, the decimal point is deleted when the digit following it is deleted.

Actually, I just noticed that the textboxes are being validated with each keypress. It seems that validation should occur after exiting the textbox.

nallath, 2015-06-05:
This could have something to do with localisation; some countries use , for decimals, others use .

BagelOrb, 2015-06-12:
I don't think so. Just became annoyed with this bug myself. It is probably a Uranium bug, I'd think.

Problem:
Enter a setting field which say "0.4" for example. Put cursor at the end of the field. Do a backspace. (Don't hit enter or anything)
Setting field becomes "0"

Expected behaviour:
Setting field becomes "0."

nallath, 2015-06-12:
This is an Uranium bug yeah.

awhiemstra, 2015-06-25:
This has now been fixed.

closed 2015-06-25 by awhiemstra