Ultimaker/Cura #28: [OS X] hidpi difficult to rotate

ErikDeBruijn, 2015-06-05:
As a user, when trying to rotate the object, I'd like it to be clear and easy to do this. It's pretty hard to see the highlight on hidpi (e.g. Retina) screens, because the circles are very thin (single pixel). Also, grabbing the line is equally hard.

If the lines would be thicker that would really make things easier.

[image]
Tested with OS X 10.10.3 

How to test:
 1. Find a retina Macbook
 1. Load .63 version of Cura
 1. Load a model
 1. Click on rotate
 1. Get the mouse exactly where in should be (this is the hard part)

nallath, 2015-06-05:
Could you use the template as found on https://github.com/Ultimaker/Uranium/wiki/Bug-Reporting-Template next time? This makes it a lot easier for us to quickly scan through issues.

ErikDeBruijn, 2015-06-06:
Will try whenever I have more time. This was in between being meetings.

awhiemstra, 2015-06-24:
So in the upcoming build I have improved the rendering of lines to account for screen DPI. On my hi-dpi testing machine that fixes this issue, but please do test once the new build (.96) is available.

BagelOrb, 2015-07-03:
I suppose the issue is fixed now.

closed 2015-07-03 by BagelOrb