Ultimaker/Cura #32: Cura's main window is blank

derekhe, 2015-06-07:
two env has same problem: 
1.Mac Yosemite
2.Ubuntu 15.04 (in virtualbox)

When opening Cura's main window, the main window is blank even after loading stl file.
There is one error is showing when click inside the window.

FileNotFoundError: Could not find resource ('default.vert',) in 5
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/site-packages/UM/Qt/Bindings/MainWindow.py", line 120, in _render
    renderer.beginRendering()
  File "/usr/local/lib/python3.4/site-packages/UM/Qt/QtGL2Renderer.py", line 154, in beginRendering
    self._initialize()
  File "/usr/local/lib/python3.4/site-packages/UM/Qt/QtGL2Renderer.py", line 321, in _initialize
    Resources.getPath(Resources.ShadersLocation, "default.vert"),
  File "/usr/local/lib/python3.4/site-packages/UM/Resources.py", line 50, in getPath
    raise FileNotFoundError("Could not find resource {0} in {1}".format(args, type))
FileNotFoundError: Could not find resource ('default.vert',) in 5

derekhe, 2015-06-07:
Solved by copy the resource from Uranium's directory. 
The window now is filled with gray color, but import the model still nothing happened.

nallath, 2015-06-08:
Virtual box has a lot of problems with openGL rendering. This is probably a problem with your graphic drivers.

derekhe, 2015-06-08:
@nallath I don't think so. I tried directly on Mac Yosemite but still can't work.

awhiemstra, 2015-06-08:
Sounds like an issue with resource lookup. There are a bunch of locations that get searched for the resource files, mostly dependant on the application binary path, which might not always be correct. Off the top of my head, at least the following are searched:

- $appdir/../Resources
- $appdir/../share/uranium
- $appdir/../share/cura
- $appdir/..

Where $appdir is the directory of the application binary. In addition, paths relative to the python files are added, but only when Python's sys module does not have the "frozen" attribute, since that usually implies a py2app/py2exe application.

Now, it is quite possbile that $appdir is incorrect in your case, I have only tested it on MacOSX using the py2app version.

derekhe, 2015-06-11:
Fixed, I added the build script here: https://github.com/Ultimaker/cura-build/pull/1

closed 2015-06-11 by derekhe