Ultimaker/Cura #33: Setting inheritance and default values

BagelOrb, 2015-06-08:
When a setting inherits (via some function) from its parent, but it's default value is different than that, which value is used? The default value or the value computed from the inherit function?

nallath, 2015-06-08:
I'd assume the inherited value.

BagelOrb, 2015-06-08:
I would say the default value, but which one is it?

awhiemstra, 2015-06-08:
If you look at: https://github.com/Ultimaker/Uranium/blob/master/UM/Settings/Setting.py#L304

First, we check to see if the setting is visible. If it is, all the inheritance logic is skipped. If it is visible, we perform a bunch of checks and effectively set the current value to the inherited value. Note that this means that if you hide a previously visible setting, the setting's value will be overridden by the inheritance code.

Once done with the inheritance code, we check to see if the current value is a valid value. If not, we return the default value. If it is, we return that value.

So, to summarize:

- If the setting is visible, we return the value of the setting or the default if no value was set.
- If the setting is not visible, we perform inheritance and override the local value, then return it or the default.

This means that inherited value takes precedence over everything, but only if the setting is not visible.

BagelOrb, 2015-06-08:
Ok, thanks.

Is that what you would want/expect, though? Supose someone sends you a file with their ubersettings, but because you don't have all settings visible, they fall back to their defaults.
I would think that a setting which has been altered remains unchanged while you hide the setting.

awhiemstra, 2015-06-08:
The tricky thing here is what the expected behaviour is. If you have someone else's settings, then yes, you would expect those settings to keep their values. Luckily the visible state is included in the saved setting configuration so exchanging those files should keep the values. In addition, sharing these settings is only partially supported at the moment, we would need a proper sharing feature to handle things like this correctly.

The more tricky thing comes from the following: I show a setting, I change that setting, then hide it. In this case, one can argue that the set value stays the same. However, what happens when I change the parent value? Or a different value that the setting depends upon? We have no real way of notifying the user that now we do override the value. Keeping the value would mean the setting has potential incorrect values. And again, we have no way of notifying the user about this.

So the most predicatable solution here is to always inherit if not visible and use the set value if it is. The only thing we may be doing wrong at the moment is that we return the set value if the setting does not have an inherited value and is not visible, instead of returning the default.

BagelOrb, 2015-06-08:
Compare with the following: I change a setting, not hide it, and then change some setting it depends upon. It might still have potentially incorrect values. If this would be different when settings are hidden, then the user is confronted with the fact that stuff changes when they are not looking.

I think that hiding something should never change it. Hiding should only hide.
Sneakily changing something when someone is not looking can make him very confused!

The next question is: when/how do we fall back to inheritance? I haven't really thought about it, but just to give you an example solution: have a separate visible-checkbox and inherit-checkbox.

BagelOrb, 2015-07-08:
The question has been answered, and the discussion is continued in the above referenced issue.

closed 2015-07-08 by BagelOrb