Ultimaker/Cura #36: Build instructions

CoDanny, 2015-06-09:
Hi,

I'm new to Cura Development. The Readme file has no instructions on environment setup or build instructions.

I'm currently monstly interested in Mac instructions. Are there somewhere else? what am I missing?

nickthetait, 2015-06-09:
This might be what you need: https://github.com/Ultimaker/cura-build

CoDanny, 2015-06-09:
@nickthetait It's possible ;) But there are no instructions there either ;)

nickthetait, 2015-06-09:
Yeah, I'm sure they will show up in a few weeks. Ultimaker is busy with making a stable release right now...

daid, 2015-06-09:
Note that the currently released Cura is at:
https://github.com/daid/Cura

Which is in maintaince mode only, due to total rework of the GUI, which is contained in this repository.

derekhe, 2015-06-13:
@CoDanny I wrote a build script for ubuntu 15.04 
https://github.com/Ultimaker/cura-build/blob/master/ubuntu-15.04-build-script.txt

CoDanny, 2015-06-15:
@daid, that's exactly why I want to work on a fork for the "future-proof" Cura. I'm trying to the 'cura-build' repo, but I'm also missing instructions for setting a development environment for the GUI project.

The goal of the fork to integrate AstroPrint into Cura for wireless printing after slicing.

awhiemstra, 2015-06-15:
@CoDanny for MacOSX, you need the latest XCode and the latest CMake. XCode is available from the Mac store or other places, CMake from www.cmake.org . Then clone the cura-build directory and run mkdir build, cmake .., make . This should download and build all needed dependencies. You can work from the subdirectories CMake creates or from separately cloned repositories, provided you do the right python setup.

Alternatively, install python 3, Qt, PyQt5 manually and build the rest of the dependcies from that.

CoDanny, 2015-06-15:
@awhiemstra Thanks for the info :)

Also, I realize that you might not be focused on instructions just yet. I might need to dig around and figure it out. Just as information as how this is perceived from the outside: These instructions should (at some point) be added to the README file. Also sentences like "provided you do the right python setup" provide little insight in how to actually achieve the goal ;)

I might have gotten involved in this project a bit too early and have to do a lot of digging on my own and that's ok. Just wanted to let you know of the holes that I see when I approach the project for the first time.

awhiemstra, 2015-06-15:
That's fine, it is valuable feedback. We know there are several bits and pieces that miss documentation, on the other hand, it is hard to document things without knowing what people need documented, since we have been looking at these things for a while now and will miss obvious stuff that we take for granted. So please do point out missing bits, even if we might not have answers directly we will at least have a list of things people need.

(And yes, I realise build intstructions are a fairly basic thing to add, those are mostly missing because we did not get around to it.)

johntron, 2015-06-18:
@CoDanny - it looks like they're using the same software as [CuraEngine](https://github.com/Ultimaker/CuraEngine), so [those instructions](https://github.com/Ultimaker/CuraEngine/blob/master/README.md#how-to-install) may help.

BagelOrb, 2015-06-19:
@CoDanny I think AstroPrint should be a plugin, right? Perhaps you don't need to fork at all! 

The new plugin structure allows for a very wide variety of plugins.

BagelOrb, 2015-06-19:
What functionality do you want to integrate?

CoDanny, 2015-06-19:
Print wirelessly, import settings from cloud, save sliced files to cloud and import designs from cloud.

Plugin would be ideal. I just don't understand enough yet about the new architecture. I can't even build it yet ;)

awhiemstra, 2015-06-19:
That should be a plugin yes, that's what they are for. Unfortunately the bit of API you need for that is rather... incomplete at the moment. The good news is though that I will start looking into that next week.

closed 2015-07-08 by nallath