Ultimaker/Cura #42: An object's "ghost" remains visible after deleting

ErikDeBruijn, 2015-06-13:
When I delete an object through the context menu, the controls (e.g. scale elipses or arrows) and a wire mesh stay visible until I click somewhere in the 3D viewport.

Where: Main interface

Actions: Using the 15.05.94 Darwin release.

Expected behavior: 
 - I would like to see the object and controls disappear entirely.

Screenshot:
[image]

Specs: OS X 10.10.3

nallath, 2015-06-15:
This is a known issue. I've also had this problem but posted it on Asana.

awhiemstra, 2015-06-25:
This is now fixed. Seems I again failed to reference the right issue though.

closed 2015-06-25 by awhiemstra