Ultimaker/Cura #43: Settings lost between Cura sessions

Xisle, 2015-06-14:
Application Version: 10.05.94
Platform: OS X 10.9.x

Steps to Reproduce:
Change settings in Cura, close Cura, reopen Cura

Actual Results:
Settings changes are lost

Expected results:
Previous settings would be retained between sessions

Ghostkeeper, 2015-06-15:
This seems to be platform-specific. It doesn't happen on my Windows 7 computer. Settings are properly retained.

awhiemstra, 2015-06-15:
Can you check if the directory .cura exists in your home directory and if so, if that contains a file cura.cfg and a directory settings?

Xisle, 2015-06-15:
[image]

Xisle, 2015-06-15:
Above is the complete contents of cura.cfg from .cura folder

Xisle, 2015-06-25:
Settings are still lost between sessions on 15.05.96.

awhiemstra, 2015-06-25:
So just to narrow things down, which settings specifically are not restored properly? Because your config file suggest at least window size and recent files are saved properly. Can you check if those are restored properly?

Xisle, 2015-06-25:
Size is maintained, but not location. That wasn't actually my issue.

The Advanced Settings are lost between sessions. Old Cura allowed these settings to be saved as a Profile, but also maintained settings between sessions.

Open Cura, change layer height or any other setting, close Cura
Open Cura... Advanced Settings revert to "default" values.

Xisle, 2015-07-01:
This appears to be fixed. I DO now see settings in my ~/.cura/settings/*.cfg file and they are retained between sessions.

I did not fully test yet, however.

Issue #76 is still an issue for me though.

closed 2015-07-08 by nallath