Ultimaker/Cura #46: Intermediate level

ErikDeBruijn, 2015-06-15:
Right now there's an beginner and advanced level. An intermediate level would also be useful for people who want just a bit more than the Beginner mode.

From the Ultimaker Evening

BagelOrb, 2015-06-15:
In the new Cura, the advanced mode doesn't automatically show all settings, so it is in some sense already the intermediate level.
You can unhide all settings via Prefereces...

awhiemstra, 2015-06-16:
The actual request was to have some sort of visibility presets so that people can choose "basics only", "everything" etc. This is a new feature that will need to be developed on top of the current systems.

maukcc, 2017-05-13:
We really like that there is only 2 options (less is more, for users)
1 recommended -> with all settings correct for a safe print, made by the vendor of the machine
2 custom -> dive in, head first, all warranties off, you are on your own :)

We would like to have the option in our definition file that would let the user see all options. Meaning that all visible settings are checked in preferences by default, when choosing Cartesio as the machine.

BagelOrb, 2017-07-21:
@ Devs, see CURA-3710

Appesteijn, 2018-02-19:
Implemented in master

closed 2018-02-19 by Appesteijn