Ultimaker/Cura #49: Automatic placement correction on/off

ErikDeBruijn, 2015-06-15:
As a user I'd like to have automatic placement be something that I can turn of, so I can add an object and see a conflict without the system automatically try to resolve it. Even better is that it always finds a way to fit everything.

nallath, 2015-06-16:
You mean the push free of the platform? 

Finding a way to fit everything is not really feasible on the python side. It would take quite a bit of computation to get that working.

BagelOrb, 2015-06-16:
Yes, we should be able to turn of the push free functionality.

awhiemstra, 2015-06-16:
The original request was for three modes: Completely off, Indicate things that are colliding using a different colour or some other indicator, but do nothing else and finally push things apart, the way it is now. On/Off switching should be a fairly simple preference to add after the release. Colouring is a little more involved as that would need changes on the view side.

BagelOrb, 2015-06-16:
I think the different colour should be the grey colour which is used to show that an object is unprintable cause it is outside the printable boundary.

I don't think the original request is for three modes. There is no need for a mode which doesn't push things around and which doesn't show which models are colliding. Or did you mean something else for the third view?

nallath, 2015-06-16:
Or simply make objects grey if the collide.

BagelOrb, 2015-06-16:
That's what I said.

nallath, 2015-08-20:
15.10 will have a button in the general preferences that allows you to disable push free.

commit bbf0b92f (2015-08-20): Added preference to disable automatic push free

fix #49
closed 2015-08-20 by nallath