Ultimaker/Cura #54: interface v2. Proposal Save toolpath button

mennoberg, 2015-06-16:
[image]
[image]

awhiemstra, 2015-06-16:
What does the toggle button do?

Also, I would prefer to avoid the term "toolpath". I think using "Save to ..." is clear enough.

mennoberg, 2015-06-16:
[image]
[image]

mennoberg, 2015-06-16:
Save button function's in this environment

[image]

closed 2015-06-25 by TMHogenhout

BagelOrb, 2015-06-29:
@awhiemstra I think it's good to explicitly disambiguate between saving toolpath and saving the model. Peerhaps your problem is woth the word 'toolpath'? Perhaps just replace the term with 'gcode'?

commit 27de7484 (2015-06-24): Restyling of the save to toolpath button

Fixes #54