Ultimaker/Cura #55: Can't see retraction settings in cura

mvangils, 2015-06-17:
Application Version: << 15.05.93>>
Platform: << Win7 64bit >>
Qt: << The version of Qt used >>
PyQt: << The version of PyQt used >>
Display Driver: << intel graphics 8.15.10.2559 >>

Steps to Reproduce:
1. go to preferences  select enable retraction
2. close propperties

Actual Results:
the retraction settings do not show in the settings bar in cura

Expected results:
retraction settings should show up in the settings bar in cura

Additional Information:

BagelOrb, 2015-06-17:
Note: For an Ultimaker 2 the Material settings are hidden.

Expected behaviour: Retraction settings should be part of another category

Actual behaviour: Retraction settings are part of the materials category

Steps to reproduce: in, out, repeat

BagelOrb, 2015-06-17:
Settings have already been refactored

awhiemstra, 2015-06-17:
This is actually relevant for the 15.06 release. The solution is to not hide the entire Material category but just hide those settings that should not be set from within Cura, like temperature.

BagelOrb, 2015-06-17:
Shouldn't you just move the setting, like I did?
It seems a bit overkill to invent a solution only for the 15.06 branch.
I can do it if you want...

awhiemstra, 2015-06-17:
We cannot move the setting to a new category because that would introduce a new string. So unless you have a good suggestion where to put it in the existing categories we need to come up with a solution that does not introduce a new category.

BagelOrb, 2015-06-18:
You could put it under [Quality] for example...

commit 6ef82c57 (2015-06-19): Do not hide the entire material category for UM2 but just the unused settings

Fixes #55
closed 2015-06-22 by awhiemstra