Ultimaker/Cura #59: Feature Request - Show filament usage by volume

czucker, 2015-06-19:
Would it be possible to add cm3 to the filament length and weight stats?

awhiemstra, 2015-06-22:
We can probably add it, would you want that information in addition to the existing or instead of the exisitng information?

Dim3nsioneer, 2015-06-22:
Would it be a possibility to make it flavor dependent? Flavors like the UMO would show mm and flavors like Ultigcode would show cm3...

BagelOrb, 2015-06-22:
I think this is a UX issue. It's about passing information to the user, which is not particularly concerned with the Gcode flavor.

czucker, 2015-06-22:
I would prefer that it be in addition to the existing information. So you would get grams, meters and cm3.

BagelOrb, 2015-07-01:
Update: as it is now, the engine passes the information to the frontend in mm^3. The only thing left to do is to display it.

Mitrasmit, 2015-07-03:
Is it just me or doesn't it show any information anymore, except the printing time? I really valued all stats; meters, grams, costs...Or is it hidden somewhere where I can't find it?

nallath, 2015-07-08:
https://github.com/nallath/PrintCostCalculator <- It's a work in progress, but it could have all these features.

nallath, 2015-07-31:
The PrintCostCalculator has this now.

closed 2015-07-31 by nallath

TMHogenhout, 2015-08-01:
https://github.com/Ultimaker/Cura/issues/194