Ultimaker/Cura #60: - ERROR - Unable to find backend executable

jasonworg, 2015-06-21:
My platform is OSX 10.10.3
I got a error when opening a stl file. It said 
~~~
2015-06-21 19:57:50,158 - ERROR - Unable to find backend executable: /Library/Frameworks/Python.framework/Versions/3.4/bin/CuraEngine
~~~
Actually, I have a config.cfg file in dir ~/.config/cura/
~~~
cat .config/cura/config.cfg
[backend] location = /Users/jasonworg/Works/Ultimaker/CuraEngine/build/CuraEngine
~~~
I found it is because of file $Cura/plugins/CuraEngineBackend/CuraEngineBackend.py
~~~
default_engine_location = os.path.join(Application.getInstallPrefix(), "bin", "CuraEngine")
        if hasattr(sys, "frozen"):
            default_engine_location = os.path.join(os.path.dirname(os.path.abspath(sys.executable)), "CuraEngine")
        if sys.platform == "win32":
            default_engine_location += ".exe"
~~~

awhiemstra, 2015-06-22:
Hmm, that's how we set the backend on the Linux side, I have never tested it on the Mac OSX side as so far I have just built packages for that and used the package build system to do any changes needed for OSX. Just to double check, is the lack of a newline between [backend] and location a paste error? Because I can imagine python's config parser ignoring the line if it is not.

jasonworg, 2015-06-22:
Hmm, I modified the config.cfg to:
~~~
[backend]
location=/Users/jasonworg/Works/Ultimaker/CuraEngine/build/CuraEngine
~~~
But the issue is still here:
~~~
2015-06-22 17:13:56,905 - ERROR - Unable to find backend executable: /Library/Frameworks/Python.framework/Versions/3.4/bin/CuraEngine
~~~
It seems that Cura didn't read config.cfg...

Eventually I add a line in CuraEngineBackend.py to solve this issue...
~~~
default_engine_location="/Users/jasonworg/Works/Ultimaker/CuraEngine/build/CuraEngine"
~~~
~~~
2015-06-22 17:26:34,765 - INFO - Started engine process: /Users/jasonworg/Works/Ultimaker/CuraEngine/build/CuraEngine
2015-06-22 17:26:34,782 - DEBUG - Backend connected on port 49674
~~~

nallath, 2015-07-08:
@jasonworg  Is this still an issue with the latest version?

closed 2015-08-14 by nallath

victor-gonzalez, 2016-01-09:
It seems README.md is wrong:

it says: 
  Link your CuraEngine backend by inserting the following line in home/.config/cura/config.cfg

but it should say:
  Link your CuraEngine backend by inserting the following line in home/.config/cura/cura.cfg

As was said it seems Cura don't read config.cfg.
At least it worked for me in OpenSUSE 13.1!