Ultimaker/Cura #61: Enum Settings and Setting Units Untranslatable

awhiemstra, 2015-06-22:
Currently, we cannot translate Enum values like support type (Touching Buildplate, Everywhere). This is due to directly using the current values in the engine. Since we do not want to have to translate everything in the engine, we should change the enum setting to use a dictionary of value => user-visible string so that we can change the visible string without impact on functionality.

In addition, we also should make units translatable, since different languages might want to use a different character set to display them.

BagelOrb, 2015-06-22:
Note that we don't have to change the enums in the engine, the engine can still work with the same strings.
The solution should just be something like changing 
```
display("enum_val")
```
to
```
display(translate("enum val"))
```

awhiemstra, 2015-06-23:
Except that that still means there is a dependency on the display string, which I also want to get rid of since it means changes in the source string cannot be done without changing the engine. I would rather do one change now to fix that instead of having to do that every time in the future.

BagelOrb, 2015-06-23:
I think that is a consequence of the fact that English is no translation really. Otherwise you could just add a translation from "enum_val" (engineering language) to "Enumeration Value" (English) for example.

Why don't you just make a proper English translation?

awhiemstra, 2015-06-23:
It has nothing to do with that. Depending on user-visible strings for these values is just wrong and will only cause us more trouble in the long run.

nallath, 2015-08-24:
Did we change anything on this regard?

awhiemstra, 2015-08-24:
Not yet. It would still be nice to get this done before 15.10 is branched.

awhiemstra, 2015-09-12:
This has now been implemented and enum labels are properly translatable.

closed 2015-09-12 by awhiemstra