Ultimaker/Cura #62: Remove limit of 3000 on Density (kg/m3) preference

czucker, 2015-06-22:
Metal infused filaments can exceed this limit, making it impossible to get accurate printing stats. BronzeFill, for instance, is 3.99 g/cm3.

It would be great if the 3000 limit was either removed, or increased to account for newer filaments.

awhiemstra, 2015-06-25:
Since we do not have such a setting currently this bug is invalid for this Cura.

closed 2015-06-25 by awhiemstra

czucker, 2015-06-25:
See the attached image for the setting that I'm speaking about:
[image]

czucker, 2015-06-25:
I hadn't realized that this setting was not present in 15.06. Will there be a place to enter filament density? If not, then how can filament volume be calculated?

nallath, 2015-06-26:
Density has nothing to do with volume. Density is used to calculate weight (by multiplying it by volume).