Ultimaker/Cura #63: Crash when opening Web Browser

nickthetait, 2015-06-22:
Version:
Cura 15.05.95

Specs:
Windows 8.1

Where:
Help dropdown menu

Actions:
Click either "Show Online Documentation" or "Report a Bug"

Behavior:
Cura crashes. Windows gives a popup: "Cura.exe has stopped working"
No new log messages on the command line.
Web browser is correctly opened.

Expected behavior:
Cura continues to function normally

nickthetait, 2015-06-22:
A side effect of this is any printers that I added/configured during this session get deleted.

Open cura, add printer, close cura. (This printer remains)
Open cura, add printer, trigger this bug & crash. (This printer is lost)

Dim3nsioneer, 2015-06-23:
The crash issue is known and seems to happen on Windows only.

commit d937eb7a (2015-06-23): Defer opening the webbrowser until the next run of the event loop

Opening a web browser from a signal handler connected to a menu
crashes Cura, so instead defer the call.

Fixes #63
closed 2015-06-23 by nallath