Ultimaker/Cura #68: Feature Request - Tab index for ALL settings

Xisle, 2015-06-25:
Is it possible to add tab indexes to checkboxes and drop-downs in the Advanced Settings pane?

nallath, 2015-06-25:
What do you mean by tab indexes?

Ghostkeeper, 2015-06-25:
Tab indices define the order of the elements in menus and forms. You know when you fill in a web form and you can type in a text box, then press tab to navigate the caret to the next text box without needing the mouse? That's tab indices.

In QML it seems to be defined by the KeyNavigation element, which allows you to specify an item that gains focus when the tab key is pressed if the current element has focus.

awhiemstra, 2015-06-25:
Yes, it is quite possible, but I do not know how soon we can. As @Ghostkeeper mentions, QML has the KeyNavigation property for it but we generate the list of settings dynamically, so it is probably harder than just setting it.

nallath, 2015-06-25:
Ah right, makes sense. It would be nice to have this.

BagelOrb, 2015-07-03:
Well, perhaps the drop-downs are easier to fix? I think that is also a more pressing issue.

nallath, 2015-08-24:
Fixed for next release.

closed 2015-08-24 by nallath