Ultimaker/Cura #69: Feature Request? - Allow slicing without "connecting" to printer

Xisle, 2015-06-25:
When the printer is busy printing via the old Cura, no slicing can occur.

nallath, 2015-06-25:
What do you mean by this? So if you are using the old cura (15.04) and try to slice with the new Cura, it doesnt work?

Xisle, 2015-06-25:
I edited my original post for clarity.

Xisle, 2015-06-25:
* New to 15.05.97 *

I have been testing 15.05.96 this morning and decided to try 15.05.97.
I have a UMO and UM2 added in Cura, though I only have a UMO. On Cura .96, I can slice using either printer as long as my UMO is plugged in and not in use (does not need to be turned on). On Cura .97 I can no longer slice for UM2.

Ghostkeeper, 2015-06-29:
I have no 3D printers at all here at home (or any Arduino devices for that matter) but Cura 15.05.97 slices fine for me. Results are as expected. Platform is Windows 8.1.

Perhaps this is an OSX-specific bug or only occurs if there was a printer connected in the past in some way while the program unexpectedly closed or was updated from an older version? In any case, not being able to slice if no printer is connected is a bug.

Xisle, 2015-06-29:
I tried installing a second UM2 and the Go and Extended. All sliced fine with or without my UMO plugged in to USB. The UM2 I had installed in an older beta version still does not. I will delete it.

So, I can slice without a given printer but still cannot slice for any printer while the old Cura process is active (even with printer unplugged from USB). If the printer happens to be busy printing via old Cura over USB, then the print also fails ( issue #70 ).

awhiemstra, 2015-06-30:
As it turns out, MacOSX uses a different error code for "address already in use" making the fallback for the engine connection fail. This is now fixed and it should be possible to slice with both the old and the new Cura running at the same time.

closed 2015-06-30 by awhiemstra