Ultimaker/Cura #71: change direction of z-axis when importing an STL file

Bas-van-Deursen, 2015-06-25:
Some CAD programs (for example Solidworks) have a different direction for the z-axis. When importing an STL file into Cura, the model is displayed in a different orientation than in the CAD program in which it was created. It would be a nice feature if Cura would recognize this and would do a translation on the model.

nallath, 2015-06-25:
I wouldn't know of a way to find out what axis frame a model is supposed to have (and compensate for that), save for checking header files and compensating on a program by program basis.

Bas-van-Deursen, 2015-06-25:
Does the STL file not contain information about from which software package it was exported?

-Bas

> 
> I wouldn't know of a way to find out what axis frame a model is supposed to have (and compensate for that), save for checking header files and compensating on a program by program basis.

awhiemstra, 2015-06-25:
Yes, that is what @nallath means with "checking header files" I think.

What I suggested when discussing this with @Bas-van-Deursen was to have an option to indicate what Axes to use for loading files, with the defaults of that set based on the STL header.

Bas-van-Deursen, 2015-06-25:
That would be nice!

However very low priority I think.

-Bas

> 
> Yes, that is what @nallath means with "checking header files" I think.
> 
> What I suggested when discussing this with @Bas-van-Deursen was to have an option to indicate what Axes to use for loading files, with the defaults of that set based on the STL header.

BagelOrb, 2015-06-29:
For now people can just rotate the object.

Bas-van-Deursen, 2015-06-29:
Ofcourse, that's why it's low priority

-Bas

> 
> For now people can just rotate the object.

BagelOrb, 2017-07-21:
I think this is fixed for 3MF. For STL it would be nice to have a preference for this.

Ghostkeeper, 2017-10-30:
Our project manager just removed this from our planning for being older than 12 weeks.

ianpaschal, 2018-02-27:
Labelling as `Won't Fix` as it's very old and very low priority: not really an "issue."

ianpaschal, 2018-03-16:
@Ghostkeeper Why did you re-open this? We have so many features and bugs to work on, I think we are lying to ourselves saying we'll get to it eventually. Gotta say "no" sometimes.

ianpaschal, 2018-05-22:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-22 by ianpaschal