Ultimaker/Cura #72: Crash to start

MiguelAngelLV, 2015-06-26:
After update, when I try start the app crash

```bash
link@Hyrule:~/dev/Cura$ ./cura_app.py 
Traceback (most recent call last):
  File "./cura_app.py", line 8, in <module>
    app = cura.CuraApplication.CuraApplication.getInstance()
  File "/usr/lib/python3.4/dist-packages/UM/Application.py", line 326, in getInstance
    Application._instance = cls()
  File "/home/link/dev/Cura/cura/CuraApplication.py", line 54, in __init__
    super().__init__(name = "cura", version = "master")
  File "/usr/lib/python3.4/dist-packages/UM/Qt/QtApplication.py", line 48, in __init__
    super().__init__(sys.argv, **kwargs)
  File "/usr/lib/python3.4/dist-packages/UM/Application.py", line 67, in __init__
    preferences.readFromFile(Resources.getPath(Resources.PreferencesLocation, self._application_name + ".cfg"))
  File "/usr/lib/python3.4/dist-packages/UM/Preferences.py", line 55, in readFromFile
    self._loadFile(file)
  File "/usr/lib/python3.4/dist-packages/UM/Preferences.py", line 120, in _loadFile
    if self._parser["general"]["version"] != "1":
  File "/usr/lib/python3.4/configparser.py", line 937, in __getitem__
    raise KeyError(key)
KeyError: 'general'
```

I use Debian, I install using Ubuntu script.

awhiemstra, 2015-06-30:
It looks like your setting file lacks a version value. You can either add it to the configuration file manually or completely remove the file and allow Cura to recreate it. You can find the file in ~/.config/cura

MiguelAngelLV, 2015-07-01:
Works now. Thanks.

closed 2015-07-01 by MiguelAngelLV

justM4D, 2015-07-01:
I have a similar error on Windows 7.
There are folders and configs for all other versions, but not for the new one, so I can't delete it.

```
Traceback (most recent call last):
  File "cura_app.py", line 8, in <module>
  File "c:\dev\uranium\UM\Application.py", line 326, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 64, in __init__
  File "c:\dev\python34\lib\configparser.py", line 937, in __getitem__
    raise KeyError(key)
KeyError: 'general'
```