Ultimaker/Cura #73: Feature request: Add view modes to menu bar

Ghostkeeper, 2015-06-29:
I think it'd enhance usability to add the view modes to the menu bar at the top of the screen, like so:

    File | Edit | View | Machine | Extensions | Settings | Help
                |   Layers |
                | * Solid  |

This option can already be changed by clicking the button with the eye in the top-right of the viewport but if an inexperienced user is searching for ways to better see the innards of his model, the menu bar is commonly the first place he'd look (even though it's not hard to find now). It seems like a simple addition to make.

If more view options would be added later on, such as a 'camera reset' button or something, that menu would also be a nice place for it.

nallath, 2015-06-29:
+1 
I think this is a good improvement.

BagelOrb, 2015-07-03:
+1
I think this is a good improvement.

commit a82717f6 (2015-07-31): Added views to top bar

Closes #73 and CURA-38 #close
closed 2015-07-31 by nallath