Ultimaker/Cura #74: Memory for Cura application window not cleared before being displayed

Xisle, 2015-06-29:
Application Version: 15.05.97
Platform: OS X 10.9.5
Video: Radeon HD 6xxx, Device ID: 0x6759

Steps to Reproduce:
Launch new Cura

Actual Results:
Application window shows previous contents of graphics memory used

Expected results:
Solid background while application finishes loading assets

nallath, 2015-06-29:
Could you add a screenshot?

Xisle, 2015-06-29:
[image]

Xisle, 2015-06-29:
[image]

Xisle, 2015-06-29:
I launched Cura twice to capture screenshots. The first time was right after closing it, the second time I waited 10 seconds and switched tasks a couple of times before reopening it (just to see the change).

BagelOrb, 2015-07-03:
This is art! I like it! :P

nallath, 2015-08-14:
Is this still an issue in the latest release?

Xisle, 2015-08-14:
yes

nallath, 2015-08-17:
It's already quite late in the process. The splashscreen is already in front of it, so is the plugin / machine loading. 
So far you've been the only one with this bug, so i think this is not a fundamental issue, but a very specific one to your system.

Xisle, 2015-08-17:
Fair enough.

nallath, 2015-08-17:
Does this happen on every startup or just every so often?

Xisle, 2015-08-17:
Every startup.

This issue can definitely be put on low priority if no one else is experiencing it.

awhiemstra, 2015-08-17:
Well, I have some ideas about improving this since the issue seems to be that the interface loading happens while the window is already shown. If we can delay showing the window until more stuff is loaded things might be better. I have also seen something similar on Windows, though it is only OSX where there is actual garbage in the window.

Xisle, 2015-09-04:
Same on 15.09.8

nallath, 2015-09-04:
Ugh :(

Xisle, 2015-09-04:
No worries. Low priority.

closed 2015-09-15 by nallath