Ultimaker/Cura #75: there are no shortcuts for people with an atypical mouse

TMHogenhout, 2015-06-29:
Application Version: 15.05.94
Platform: Ubuntu 15.04 64-bit
Qt: 5.4.1
PyQt: 5.4.1
Display Driver: xserver-xorg-video-intel 2.99.917

Steps to Reproduce:
- Try to navigate the platform with a mouse that only has one or two buttons (most apple mouses and older mouses), instead of three (usually the scrolling wheel is the third button). 

Actual Results:
- You can't zoom or rotate and in some cases you can't use the move/pan function. 

Expected results:
- With the use of shortcuts these functions are available. And the actual shortcuts (ctrl-click for example) are known to the user.

TMHogenhout, 2015-06-29:
Shortcut proposal. The shortcuts are chosen on the next criteria: 

1. What kind of navigational pattern was used by Cura in the past? 
2. What kind of navigational pattern is used on CAD-programs that are commonly used by the target group? When the navigational patters between CAD and CAM are roughly the same, this ensures a smooth navigational transition between these programs. It turns out that the programs that are being used are very diverse and so are their navigational patterns. But for the sake of: the only thing worse then a bad decision is no decision at all. So here it is: 

[image]

daid, 2015-06-29:
Current cura does:

* Zoom: scroll wheel, "[+]/[-] with [shift]", [numpad +] [numpad -], [mid mouse drag], [left + right mouse drag]
* Rotate view: [right mouse drag], [numpad arrow keys]
* Pan view: [right mouse drag]+[shift]

BagelOrb, 2015-07-03:
Indeed the best way to solve the issue is not to have different shortcuts per mouse, but just have multiple shortcuts per function, of which at least one only requires the left mouse button at most.

TMHogenhout, 2015-07-07:
Short-cuts will be added to the Manual

closed 2015-07-07 by TMHogenhout