Ultimaker/Cura #76: Settings lost when switching between Advanced and Simple settings

Xisle, 2015-07-01:
Application Version: 15.06
Platform: OS X 10.9.5

Steps to Reproduce:
Change something in Advanced Settings
Switch back to Simple Settings
Switch back to Advanced Settings

Actual Results:
Advanced Settings changes are lost

Expected results:
Cura would retain settings

awhiemstra, 2015-07-01:
I tried this on my local OSX install and there settings are retained. Are any files created in ~/.cura/settings ? Additionally, if you start through the command line using <Install Dir>/Cura.app/Contents/MacOS/Cura , do you get any errors?

Xisle, 2015-07-01:
[image]

Xisle, 2015-07-01:
[image]

Xisle, 2015-07-01:
Launching from command line produces this:

[image]

Xisle, 2015-07-01:
This issue could be related to issue #43

nallath, 2015-08-14:
I have the same issue; Switching back somehow forces the defaults.

nallath, 2015-09-15:
Fixed for 15.10

closed 2015-09-15 by nallath