Ultimaker/Cura #77: Separate setting inheritance from visibility

BagelOrb, 2015-07-01:
(See also https://github.com/Ultimaker/Cura/issues/33 for a general discussion on setting inheritance)

As it is now, changing a setting to invisible alters the value. Toggling its visibility also toggles whether it inherits.

For a novice user this might come as a big surprise. One would not expect a setting to change based on whether it's visible. Also because it is not visible the user will have trouble finding the cause of the change in his print when it's done.

An advanced user might want to tweak the values of a lot of settings a bit from their initial values. After that he wouldn't want to see all those settings any more. He shouldn't be forced to keep them all visible. He will only be tweaking a couple of settings like layer height and wall thickness. It should be possible for him to hide the other tweaked settings without changing their values.

As I see it the following states on a setting should be introduces/separated:
- visible, not inherited (a.k.a. visible)
- invisible, not inherited (a.k.a. locked)
- invisible, inherited (a.k.a. inherited)
- NOT [visible, inherited] !

Note that when all children are locked, their parent should not be able to be visible.

Separating inheritance from visibility can also lead to the solution of a couple of other problems. Suppose one user saves it's settings for sharing. I would suppose then only the visible settings are saved. Another person loading these settings then doesn't need to have the same settings visible and loading the settings doesn't have to change the visibility of the settings. Instead all saved settings should become locked, when loaded. (There are more problems to this use case, but at least the visibility problem is solved.) The proposed solution allows for presets to be loaded on top of each other; one preset could cover only shell settings while the other covers only speed settings.

Suppose I edit the infill percentage and then make the line distance visible. Because the infill percentage gets invisible, its value is inherited/derived, so one might expect the infill line distance to be inherited/derived via via from the parent of the infill percentage. However, because we just adjusted the infill percentage, one would hope the infill line distance would be derived from the infill percentage. I don't know what's the expected or actual behaviour, but this doubt is caused only by the fact that changing the visibility of the child means changing the visibility of its parent which in turn means changing the value, which means changing the value of the child setting. The doubt is absolved when visibility and inheritance are decoupled.

BagelOrb, 2015-07-08:
I think this is high priority. It will take users a while before they notice something is wrong, so you might expect such a report from a standard user a bit later.

What do you think, @awhiemstra ?

nallath, 2015-08-24:
Fixed for next release

BagelOrb, 2015-09-29:
I don't understand how this is fixed now. I think the original issue still stands. 

If the original issue is solved then making a setting (in)visible should ***never*** change behaviour. It shouldn't change whether the setting has an inherited value or not.

@nallath can you explain in what behaviour is new which you thought solved the issue?

BagelOrb, 2015-09-29:
See also https://github.com/Ultimaker/Cura/issues/78

nallath, 2015-09-30:
The settings now have an inheritance value, which sets the value of a setting regardless of what it's parents are. 
As far as I can see, all settings that inherit values /visibility from each other (such as linewidth, wall line width, outer / other wall line width) it makes total sense to have both the visibility and the inheritance coupled.

BagelOrb, 2015-09-30:
There is a syntax error in the post above; I cannot parse it.

I still don't understand what you have changed which you would call separating inheritance from visibility. Please explain.

I will clarify what I mean.

Inheritance: whether the value of a setting is computed from the inheritance function instead of taken from user input

Visibility: whether a setting is present in the list on the right hand of the screen.

Separation/Decoupling: I am talking about separating inheritance and visibility from *each other*. 

The checkbox in the setting preferences menu currently performs both an inheritance and a visibility change: making a setting visible and non-inherited by checking it. Making a setting invisible by unchecking the checkbox can currently make it inherited or  non-inherited, depending on whether the user has changed the value from it's default.

Proposed way of decoupling visibility and inheritance
----
When checking the checkbox next to a setting in the preferences the setting only becomes visible, but stays inherited. The visual counter part of this could be that the text box is grey.

Then the setting could be made non-inherited in some way. For example clicking the grey text box makes white and editable. Also an icon button appears by which you can make it inherited again. Currently something very similar is done by displaying a refresh arrow. Instead maybe a link icon is more appropriate. 
[image]

This way one can see a visible inherited setting change as you change other settings. This is a way of making the inheritence relationships clear to users which cannot easily parse the json files off the tops of their heads.

===========================================================

Please explain what you have changed which you would call separating inheritance from visibility.

nallath, 2015-09-30:
I didn't change anything, Arjen did. I misunderstood your point, but now I've read your extended explanation I disagree with a full decoupling of visibility and inheritance. I think the current behaviour makes a lot of sense. If a certain setting is no longer visible it's value should no longer influence any parameter. 

Adding an extra icon to indicate inheritance only complicates the current work flow. The inheritance is already much clearer in the new interface.

BagelOrb, 2015-09-30:
It's not an extra icon, since it would just replace the refresh arrow next to a changed setting.

Currently it's unclear to me how exactly the reset to default button would work. Does it retrieve the default value or the inherited value? I should hope the latter; otherwise the default value of a line width setting would be 0.4 even if the nozzle size is 0.6mm.

You say that the inheritance is much clearer in the new interface, but I don't see ***any*** indicator of inheritance. I cannot derive any inheritance relationship. I think you are thinking of the parent/child relationship, which is principally a different thing. Whether an inheritance function uses the parent_value is as arbitrary as it using any other value. 

You say that if a certain setting is no longer visible its value should no longer influence any parameter. The opposite is now the case. When a setting is invisible it can either have a user defined value or an inherited value, based on whether you've changed the setting. Either way it can still influence other settings with inherited values if their inheritance functions refer to it.

nallath, 2015-09-30:
[image]
I think this is a pretty clear indication.

Settings should default when they are invisible. If they don't, that's a bug and that should be fixed (instead of changing how everything works ;))

BagelOrb, 2015-09-30:
No, settings should get their inherited values, not their default values.

As I said, that is an indication of parent/child relationships which is separate from inheritance relationships. personally I think the parent/child relationships are a bit superfluous. It's only used to decide that a parent should be disabled if all children are enabled (which is bugged right now).

nallath, 2015-09-30:
No, the child parent relation also decides some value inheritance.

BagelOrb, 2015-09-30:
That is to say that the inheritance function defaults to "parent_value" if it's not given. In the end it's always the inheritance function which decides from which settings it derives its value.

Ok some settings inherit their value plainly from their parent, but that's not always the case (e.g. infill_line_distance). 

Now we're showing the parent child relation, which indicates *some* inheritance, instead of making the inheritance itself clear to the user.

BagelOrb, 2015-09-30:
To get a setting back to the inherited state you now first have to reset it to default and then make it invisible. This is quite a weird detour.

BagelOrb, 2015-09-30:
It's also a debugging nightmare to not be able to see what the inherited value of a given setting is.

I want to see if it works that when the infill gets above a certain percentage, the infill pattern changes. You would expect that making the setting visible let's you view the value it currently has, but as soon as you make it visible it gets a different value.

BagelOrb, 2015-10-01:
There is no way to return to standard settings because visibility influences the settings and there is no way to return to standard visibilities.

BagelOrb, 2015-10-21:
@awhiemstra This is implemented on 15.10

As to UI, I think when the input box is in an inherited state (it has no refresh arrow) the box should also be  grey. That way you make clear it is not just a value like the one you entered, but a changing value.

@TMHogenhout @mennoberg How do you think the user is visually more inclined to understand that the setting value of a particular setting is inherited and changing with other parameters instead of just a static value?

BagelOrb, 2016-04-26:
Fixed

closed 2016-04-26 by BagelOrb