Ultimaker/Cura #79: Drag and Drop functionality on OS X

Xisle, 2015-07-01:
Application Version: 15.06
Platform: OS X 10.9.5

Steps to Reproduce:
Drag and drop a .stl file onto Cura

Actual Results:
Cura accepts the .stl file, but OS X shows an animation that implies otherwise (the file moves back to where it was dragged from)

Expected results:
OS X would be notified that the drag and drop succeeded to eliminate the reverse animation (just a guess)

Ghostkeeper, 2015-07-02:
It's really nice to have such extensive bug tests, Xisle :) Thank you!

Xisle, 2015-07-02:
You're welcome. When I finally get a Print button to show up, I can continue testing ;)

Xisle, 2015-08-06:
I think objects dropped into Cura should be centered on the coordinates of the drop if possible.

Xisle, 2015-08-06:
It would also be nice if the Cura icon (in the OS X tray) accepted drag-and-drop events.

awhiemstra, 2015-08-07:
I actually committed a change for the last one yesterday, though I need to check if it actually works now.

nallath, 2015-08-10:
So, does it work now?

closed 2015-09-15 by nallath