Ultimaker/Cura #80: Error when opening Cura

Cinezaster, 2015-07-01:
Application Version: 16.05
OS X version:  10.10.3

Steps to Reproduce:
I double clicked on Cura. 

Actual Results:
I get an error message. Result in console:
```
1/07/15 13:15:14,492 Cura[2359]: Cura Error
1/07/15 13:15:14,492 Cura[2359]: 2015-07-01 13:15:14.491 Cura[2359:466070] Cura Error
1/07/15 13:15:15,616 com.apple.xpc.launchd[1]: (ie.hood.local-tld-service[2373]) Service exited with abnormal code: 1
1/07/15 13:15:15,616 com.apple.xpc.launchd[1]: (ie.hood.local-tld-service) Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
1/07/15 13:15:16,223 Cura[2359]: Traceback (most recent call last):
1/07/15 13:15:16,223 Cura[2359]:   File "/Volumes/Cura-15.06.00-Darwin/Cura.app/Contents/Resources/__boot__.py", line 77, in <module>
1/07/15 13:15:16,224 Cura[2359]:     _run()
1/07/15 13:15:16,224 Cura[2359]:   File "/Volumes/Cura-15.06.00-Darwin/Cura.app/Contents/Resources/__boot__.py", line 62, in _run
1/07/15 13:15:16,224 Cura[2359]:     exec(compile(source, path, 'exec'), globals(), globals())
1/07/15 13:15:16,224 Cura[2359]:   File "/Volumes/Cura-15.06.00-Darwin/Cura.app/Contents/Resources/cura_app.py", line 8, in <module>
1/07/15 13:15:16,224 Cura[2359]:     app = cura.CuraApplication.CuraApplication.getInstance()
1/07/15 13:15:16,224 Cura[2359]:   File "UM/Application.pyc", line 326, in getInstance
1/07/15 13:15:16,224 Cura[2359]:   File "cura/CuraApplication.pyc", line 64, in __init__
1/07/15 13:15:16,224 Cura[2359]:   File "configparser.pyc", line 937, in __getitem__
1/07/15 13:15:16,225 Cura[2359]: KeyError: 'general'
```
This also happens when I first copy it to my application folder.
Tried it on an other laptop. Same result

Expected results:
That Cura opens

nallath, 2015-07-01:
We're aware of this error. Hotfix should be up soon.

awhiemstra, 2015-07-06:
This should be fixed with 15.06.01

closed 2015-07-06 by awhiemstra