Ultimaker/Cura #81: Cannot remove a saved printer in preferences>machine.

jimustanguitar, 2015-07-01:
Running Win7-64 and Cura 15.06.01. After adding a second printer, I can't remove either one. It looks right, and it asks me if I'm sure, but the one I removed is still there. 

They have the same name (created in error, human error) perhaps that's part of the difficulty.

nallath, 2015-07-01:
Try running Cura in admin mode, that should allow it to create a log file.

awhiemstra, 2015-07-01:
Looks like there currently is no way to remove machines that have been added this session. Restarting cura should fix your issue as Cura will only write one file.

The error log thing is known and happens when there is a python error for some reason or another. I will look at implementing a proper crash handler later.

closed 2015-07-10 by awhiemstra

badvision, 2017-11-11:
This bug is still happening in the current version, at least in Windows 10