Ultimaker/Cura #82: No Print option on OS X

Xisle, 2015-07-01:
Application Version: 15.06.01
Platform: OS X 10.9.5

Steps to Reproduce:
Open Cura with UMO plugged in via USB and powered up

Actual Results:
Cura checks for printer, printer makes clicking noise (as normal), after slicing no Print option is available.

Expected results:
Cura would allow me to print ;)

See screenshots in issue #76. I think they may contain useful information for this issue.

nallath, 2015-07-01:
The part that gives the error is exactly the same as the old cura. Weird.

awhiemstra, 2015-07-09:
Can you install the package pyqt5-examples and try to run some of those that relate to QtQuick?

Xisle, 2015-07-11:
Cura 15.06.02

The USB connection doesn't "fail" now.
I still have no print button though.

2015-07-10 15:06:48,488 - INFO - Loaded plugin FileLogger
2015-07-10 15:06:49,281 - INFO - Loaded plugin LocalFileStorage
2015-07-10 15:06:49,311 - INFO - Loaded plugin MeshView
2015-07-10 15:06:49,324 - INFO - Loaded plugin WireframeView
2015-07-10 15:06:49,340 - INFO - Loaded plugin LayerView
2015-07-10 15:06:49,350 - INFO - Loaded plugin OBJReader
2015-07-10 15:06:49,358 - INFO - Loaded plugin STLReader
2015-07-10 15:06:49,368 - INFO - Loaded plugin OBJWriter
2015-07-10 15:06:49,376 - INFO - Loaded plugin STLWriter
2015-07-10 15:06:49,390 - INFO - Loaded plugin GCodeWriter
2015-07-10 15:06:49,403 - INFO - Loaded plugin CameraTool
2015-07-10 15:06:49,422 - INFO - Loaded plugin MirrorTool
2015-07-10 15:06:49,507 - INFO - Loaded plugin RotateTool
2015-07-10 15:06:49,525 - INFO - Loaded plugin ScaleTool
2015-07-10 15:06:49,535 - INFO - Loaded plugin SelectionTool
2015-07-10 15:06:49,548 - INFO - Loaded plugin TranslateTool
2015-07-10 15:06:49,563 - INFO - Checking for new version of cura
2015-07-10 15:06:49,563 - INFO - Loaded plugin UpdateChecker
2015-07-10 15:06:49,610 - INFO - Loaded plugin USBPrinting
2015-07-10 15:06:49,636 - DEBUG - Attempting to connect to /dev/cu.usbmodem1a11
2015-07-10 15:06:49,646 - INFO - Loaded plugin CuraEngineBackend
2015-07-10 15:06:50,080 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-10 15:06:50,094 - DEBUG - Backend connected on port 49674
2015-07-10 15:06:53,281 - DEBUG - Loaded theme /Applications/Cura 15.06.02.app/Contents/Resources/cura/resources/themes/cura
2015-07-10 15:06:57,497 - DEBUG - Loaded a mesh with 10452 vertices
2015-07-10 15:07:16,939 - DEBUG - Loaded a mesh with 26208 vertices
2015-07-10 15:07:32,317 - DEBUG - Loaded a mesh with 10452 vertices
2015-07-10 15:07:33,035 - DEBUG - Killing engine process
2015-07-10 15:07:33,191 - INFO - Backend crashed or closed. Restarting...
2015-07-10 15:07:33,463 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-10 15:07:33,469 - DEBUG - Backend connected on port 49674
2015-07-10 15:07:36,688 - WARNING - Printer connection state was not changed
2015-07-10 15:07:36,703 - WARNING - Printer connection state was not changed
2015-07-10 15:08:33,189 - DEBUG - Killing engine process
2015-07-10 15:08:33,611 - INFO - Backend crashed or closed. Restarting...
2015-07-10 15:08:33,888 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-10 15:08:33,893 - DEBUG - Backend connected on port 49674
2015-07-10 15:08:34,873 - DEBUG - Killing engine process
2015-07-10 15:08:34,880 - INFO - Backend crashed or closed. Restarting...
2015-07-10 15:08:35,155 - INFO - Started engine process: /Applications/Cura 15.06.02.app/Contents/MacOS/CuraEngine
2015-07-10 15:08:35,160 - DEBUG - Backend connected on port 49674

Xisle, 2015-07-12:
-UPDATE on 15.06.02-

I launched Cura this morning with my printer turned off... I got a greyed out "Print to USB" button!! I turned on my printer, loaded a model and appeared to be able to print (aside from the heated bed bug, which I'm not worried about). I didn't bother printing - I wanted to see Cura launch again with a print button, but it didn't.

Apparently I only get a "Print to USB" button when I unplug/replug the USB and unplug the DC IN (can be left out, or not) on my UMO, and only sometimes. I tried cycling USB only with no luck. When I can get the button to appear, it won't appear again upon relaunching Cura.

The "WARNING - ..." line in the .log is new to my setup AFAIK.

..time to try printing some stuff :)

Xisle, 2015-07-15:
Still having issues with the "new" 15.06.02. I get no Print to USB. I have yet to be able to print anything due to this error and others...

I'm not complaining, but rather keeping you informed of issues with my hardware setup.

nallath, 2015-08-14:
Is this still an issue?

Xisle, 2015-08-14:
Not as stated, but USB printing still has issues.

-If a print is canceled, Cura loses connection with printer.
-If settings are changed while printing, the print stutters.

Xisle, 2015-09-04:
The black button in lower right for Print or Save fires no actions. File -> Save All -> Print to USB works, however.

nallath, 2015-09-04:
I fixed the black button in the lower right. Should work for next release.

closed 2015-09-15 by nallath