Ultimaker/Cura #84: Unable to Select Model in version 15.06.x

smallce, 2015-07-02:
Hello,

I've installed an early version Cura in the past (back to Feb 2015, where the latest Cura version at that point was 15.02.1) and had no problem at all on my working machine ( DELL Latitude E6540, running Windows 7 Professional Traditional Chinese Edition) . 

However, after installing latest release (15.06.1) version, I found that I cannot select any model after model importing - therefore, no model manipulating could be done. I even launched the older version (15.02.1) and run it side-by-side, and found no such problem in older version. 

So far, different versions give me different results: (all result tested on the same version, even run them side-by-side)

15.02.1 - select model works, load complex model fast
15.04 - select model works, load complex model fast
15.06.0 - select model not working, take a considerable longer time for loading complex model
15.06.1 - select model not working, take a considerable longer time for loading complex model

Side note - I try to install it(15.06.0) on my Surface Pro which running windows 8.1 pro and having no trouble select model or manipulate selected one.

Would you mind pointing it out what may cause this issue?

Best Regards,
SIN-IN HO

BagelOrb, 2015-07-03:
Some problems might be caused by the fact that you have Cura doubly installed. Perhaps this is one of those problems.

awhiemstra, 2015-07-06:
Please provide information about what video chip you are using, since it sounds like you are running on something that does not support Framebuffer Objects. Additionally, try updating your drivers since that might improve that bit of support.

nallath, 2015-07-08:
@smallce Could you please provide the requested information?

Ppyromann, 2015-07-15:
Having the same issue on 15.06.02.  I cannot select any objects loaded on the platform.  Have not tested any other 15.06.xx versions.  No issues with 15.02.RC1 and prior
Running:
Windows 7 Enterprise 64 bit
ATI FirePro V5700 (FireGL)

alonsorobots, 2015-08-03:
I'm also having this issue. I had to revert back to 15.04.2

awhiemstra, 2015-08-03:
If you can provide the output of dxdiag and other log output I might be able to figure out what causes this.

Ppyromann, 2015-08-03:
Log Files:
https://gist.github.com/Ppyromann/4e77948df7fac642de1c

awhiemstra, 2015-08-04:
Hmm:

> Driver Date/Size: 6/14/2009 21:47:38

That worries me, since that suggest the driver is over 6 years old.

According to http://support.amd.com/en-us/download/workstation/previous/detail?os=Windows%207%20-%2064&rev=8.911.3.4 the latest driver for that Chip is from 6/20/2012, please try and see if you can install that and it resolves the issues.

closed 2015-08-14 by nallath