Ultimaker/Cura #85: not printing hollow?

yzorgone, 2015-07-02:
Newest Cura 
I wanted to print a hollow object.  however i cant get cura to not fill my object.
i have set the infill parameter to 0% for this.

also i have a slight feeling that opening stl files takes somewhat longer in new Cura.
[image]

BagelOrb, 2015-07-02:
This bug is resolved in the master branch in fdmprinter.json
The inherit function for infill_line_distance was wrong in the 15.06 branch

BagelOrb, 2015-07-02:
Fixed on 15.06

closed 2015-07-02 by BagelOrb

yzorgone, 2015-07-02:
okay…  that doesn’t tell me how to fix my problem.
i get the error  Cura-15-06-01 if you got anything newer. please
can you point me the direction to the download?

BagelOrb, 2015-07-02:
It will be included in the distributed Cura soon. 

However, until then you can unhide the infill line distance setting (Machine Preferences > scroll down > tick the checkbox next to [Infill Line Distance]) and set it to zero. 

(Zero is a special case, meaning the distance between the infill lines is infinite, i.e. there are no infill lines.)