Ultimaker/Cura #87: Feature Request: Load old settings

BagelOrb, 2015-07-02:
It would be nice if people wouldn't have to reconfigure all the setting preferences they saved from the old Cura.

The new Cura should be able to read and 'translate' those files to the new settings.

loelkes, 2015-07-07:
Same question: did i miss it or is there no way to load 'old' cura profiles.

nallath, 2015-07-07:
There is no way to load old cura profiles at this moment.

loelkes, 2015-07-07:
Is is possible to export profiles?

nallath, 2015-07-07:
It's possible for the old cura, but the new one cant import them. This is because a lot of the naming in the old cura was really weird, so we had to break it.

loelkes, 2015-07-07:
OK. I'll stay with the old Cura until the new one has at least an import/export feature since i have different computer at differentes places where i use the same profile.

BagelOrb, 2015-07-07:
It is a difficult feature atm, since the values of hidden settings are now calculated from other settings values. Should loading a settings profile from the old cura mean that all included settings become visible? What about settings that are new in the new Cura (e.g. coasting, support towers, etc.), or settings which are replaced by totally different settings (e.g. retraction limitation settings)?

loelkes, 2015-07-07:
The 'import from old cura' feature would be nice but if it's not possible... but if i manually set up my new profile in Cura i would like to be able to backup it and send it to other computers or friends.

nallath, 2015-07-07:
The current values of the settings are saved in [machine_name].cfg and the location is pased on the operating system. For windows its in appdata/local/cura/settings, linux its ~/.local/share/cura/settings and mac is home/.cura/settings

loelkes, 2015-07-07:
OK, thanks, works for me. Maybe a button for less advanced users in the next version could be nice.

nallath, 2015-07-07:
Agreed. It's on the roadmap for next release.

nickthetait, 2016-04-06:
This one got fixed some time ago correct?

fieldOfView, 2016-04-06:
Yes, thanks

closed 2016-04-06 by fieldOfView