Ultimaker/Cura #88: could not create shader / coredump on 15.06.1 + Ubuntu 64 15.04

Patola, 2015-07-02:
Just downloaded Cura 64-bits 15.06.1 from http://ultimaker.com/en/products/cura-software
Installed ok on Ubuntu 64-bits 15.04 (had to uninstall libprotobuf-dev and a bunch of other -dev packages though).
When I try to run it, though, it spits out:

[2:46] [4290] [patola@ovo applications]% cura_app.py
2015-07-02 02:46:40,397 - INFO - Loaded plugin ConsoleLogger
2015-07-02 02:46:40,478 - INFO - Loaded plugin LocalFileStorage
2015-07-02 02:46:40,480 - INFO - Loaded plugin MeshView
2015-07-02 02:46:40,482 - INFO - Loaded plugin WireframeView
2015-07-02 02:46:40,484 - INFO - Loaded plugin LayerView
2015-07-02 02:46:40,485 - INFO - Loaded plugin OBJReader
2015-07-02 02:46:40,486 - INFO - Loaded plugin STLReader
2015-07-02 02:46:40,487 - INFO - Loaded plugin OBJWriter
2015-07-02 02:46:40,488 - INFO - Loaded plugin STLWriter
2015-07-02 02:46:40,490 - INFO - Loaded plugin GCodeWriter
2015-07-02 02:46:40,491 - INFO - Loaded plugin CameraTool
2015-07-02 02:46:40,493 - INFO - Loaded plugin TranslateTool
2015-07-02 02:46:40,496 - INFO - Loaded plugin MirrorTool
2015-07-02 02:46:40,498 - INFO - Loaded plugin SelectionTool
2015-07-02 02:46:40,501 - INFO - Loaded plugin ScaleTool
2015-07-02 02:46:40,516 - INFO - Loaded plugin RotateTool
2015-07-02 02:46:40,518 - INFO - Checking for new version of cura
2015-07-02 02:46:40,519 - INFO - Loaded plugin UpdateChecker
2015-07-02 02:46:40,525 - INFO - Loaded plugin USBPrinting
2015-07-02 02:46:40,531 - INFO - Loaded plugin CuraEngineBackend
2015-07-02 02:46:40,536 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
2015-07-02 02:46:40,536 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
2015-07-02 02:46:40,536 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
2015-07-02 02:46:40,536 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
2015-07-02 02:46:41,139 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/share/cura/resources/qml/SidebarSimple.qml:119:22: Unable to assign [undefined] to bool
2015-07-02 02:46:41,356 - INFO - Started engine process: /usr/bin/CuraEngine
2015-07-02 02:46:41,363 - DEBUG - Backend connected on port 49674
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( imageTexture ): shader program is not linked
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( imageTexture ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( imageTexture ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( imageTexture ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( u_ambientColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_diffuseColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_specularColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_shininess ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( u_ambientColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_diffuseColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_specularColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_overhangColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_shininess ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( u_ambientColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_diffuseColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_specularColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_overhangColor ): shader program is not linked
[... 113 lines cut ...]
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed:
""
QOpenGLShaderProgram::uniformLocation( matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QQuickCustomMaterialShader: Shader compilation failed:
""
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( qt_Matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( src ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( qt_SubRect_src ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( qt_Opacity ): shader program is not linked
zsh: segmentation fault (core dumped) cura_app.py
[2:46] [4291] [patola@ovo applications]%

nallath, 2015-07-02:
What video card do you have (and what drivers?)

Patola, 2015-07-02:
*-display
                description: VGA compatible controller
                product: GF116M [GeForce GT 560M]
                vendor: NVIDIA Corporation
                physical id: 0
                bus info: pci@0000:01:00.0
                version: a1
                width: 64 bits
                clock: 33MHz
                capabilities: pm msi pciexpress vga_controller bus_master cap_list rom
                configuration: driver=nvidia latency=0
                resources: irq:37 memory:f4000000-f5ffffff memory:e8000000-efffffff memory:f0000000-f3ffffff ioport:e000(size=128) memory:f6000000-f607ffff

[15:34] [4508] [patola@ovo patola]% dpkg -l | grep -i nvidia | grep -v '^rc'
ii  bbswitch-dkms                                               0.7-2ubuntu1                                      amd64        Interface for toggling the power on nVidia Optimus video cards
ii  libcuda1-340                                                340.76-0ubuntu2                                   amd64        NVIDIA CUDA runtime library
ii  libcuinj64-6.5:amd64                                        6.5.14-1                                          amd64        NVIDIA CUINJ Library (64-bit)
ii  nvidia-331                                                  340.76-0ubuntu2                                   amd64        Transitional package for nvidia-331
ii  nvidia-340                                                  340.76-0ubuntu2                                   amd64        NVIDIA binary driver - version 340.76
ii  nvidia-cuda-doc                                             6.5.14-1                                          all          NVIDIA CUDA and OpenCL documentation
ii  nvidia-opencl-icd-340                                       340.76-0ubuntu2                                   amd64        NVIDIA OpenCL ICD
ii  nvidia-prime                                                0.8.1                                             amd64        Tools to enable NVIDIA's Prime
ii  nvidia-profiler                                             6.5.14-1                                          amd64        NVIDIA Profiler for CUDA and OpenCL
ii  nvidia-settings                                             346.59-0ubuntu1                                   amd64        Tool for configuring the NVIDIA graphics driver
ii  nvidia-visual-profiler                                      6.5.14-1                                          amd64        NVIDIA Visual Profiler for CUDA and OpenCL

Patola, 2015-07-02:
Note that all OpenGL programs work, including Cura up to 15.04.

nallath, 2015-07-03:
Cura 15.04 used an older OpenGL.

Patola, 2015-07-03:
Just tried on another computer with Ubuntu 15.04 64 bits, same thing. I have also tried to erase the old ~/.cura settings directory to no avail. I am unable to run the newer Cura...

awhiemstra, 2015-07-06:
Unfortunately it seems the entire Qt OpenGL stack crashes for you. Can you provide what glxinfo outputs for your card?

Patola, 2015-07-07:
I am pasting the output of 2 machines where this problem occurs, both with Ubuntu 15.04 64-bit. The first one is the computer where the output I pasted to this ticket occured. 

Since github does not support non-image attachments, I am pasting it to pastebin:

Computer 1: http://pastebin.com/xDNL5ymS
Computer 2: http://pastebin.com/pLGPCAH8

awhiemstra, 2015-07-09:
Can you install the package pyqt5-examples and try and run some of those related to QtQuick?

Patola, 2015-07-10:
I can't. They keep spitting the same type of errors. Maybe it's the lack of some libraries?

QQmlExpression: Expression file:///usr/share/doc/pyqt5-examples/examples/quick/tutorials/extending/chapter1-basics/app.qml:57:15 depends on non-NOTIFYable properties:
    PieChart::name
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed: 
""
QOpenGLShaderProgram::uniformLocation( qt_Matrix ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed: 
""
QOpenGLShaderProgram::uniformLocation( matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( alphaMin ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( alphaMax ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( fontScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( vecDelta ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed:

Patola, 2015-07-10:
This is my list of packages installed. Which qt5 library may be lacking? https://gist.githubusercontent.com/Patola/ed260bf2ada194cd62f1/raw/my%20list%20of%20packages%20for%20cura%20issue%20%2388

Patola, 2015-07-10:
This seems to be related to a nvidia bug with pyqt4 / 5. Bug report here: https://bugs.launchpad.net/ubuntu/+source/python-qt4/+bug/941826

It seems that using "from OpenGL import GL" makes it work.
I tried with the sample applications, e.g. chapter1-basics from pyqt5-examples and this made it work.
I have tried to insert "from OpenGL import GL" into cura_app.py and CuraApplication.py but it didn't work (couldn't find module OpenGL).

Patola, 2015-07-10:
Hurray! I installed python3-opengl and modified /usr/lib/python3/dist-packages/cura/CuraApplication.py by adding "from OpenGL import GL" and it worked! Finally! Should I fork and make a push request?

Patola, 2015-07-10:
There you go, my first pull request ever :P https://github.com/Patola/Cura/commit/a4287925618fd722347102fdbdc40b35ccc295a0

Well, I think I'd have to also add python3-opengl to the debian control file (depends:).

Patola, 2015-07-10:
Fixed my previous pull request with this one - https://github.com/Patola/Cura/commit/e89b72b3d7ab3927c648e5dd4ab1be6aedab3af4

I don't know if it worked. I hope so. I am still new at this.

plasmasolutions, 2015-07-14:
Hi Patola,

I'm suffering from the same bug... could you tell me which packages you installed? Because with your fix, I'm getting a "no module named OpenGL"

Greetings, Thomas

Patola, 2015-07-14:
I think I installed python3-opengl for it to work. If it still does not work, try python-opengl.

plasmasolutions, 2015-07-14:
Awesome! It was exactly that python3-opengl ! That should be a dependency then. Thanks a bunch and good to have it solved finally!

closed 2015-08-14 by nallath

avsdev-cw, 2017-03-09:
On my systems (all running debian with nVidia cards/drivers) in addition to "from OpenGL import GL" you also need to have "from OpenGL import GLU" or you will get errors about missing attributes (note: import GLU before GL)

dimsplendid, 2018-05-23:
The problem still exists, but install PyOpenGL can be solved, nice!

Ghostkeeper, 2018-06-29:
I wonder if this is only a problem when running from source or whether it also happens when running the AppImage.