Ultimaker/Cura #89: Support for other printers

quillford, 2015-07-02:
Are there any plans for support for non Ultimaker printers? It would be nice if there was at least an "other" option where the user could specify their printer's properties such as build volume.

Morgul, 2015-07-02:
I was just about to file a bug with this. I have a home-build delta printer I've been using Cura with since I built it, but the new version no longer lets me specify a circular build plate with 0,0,0 as the center. I don't need much, just that. Specifying the build volume would be nice to have, but I could live without. As it stands right now, though, delta printers are simply not supported.

Xisle, 2015-07-02:
Check out this topic..

https://ultimaker.com/en/cura/view/16409-thoughts-about-the-new-cura?page=8

Morgul, 2015-07-02:
@Xisle I attempted to add a new .json file, but Cura doesn't pick up on it. 

Additionally, while it has setting 0,0,0 as center, it does not seem to have any way to specify the build plate as circular. Not to mentions, some of those settings are meaningless to anyone but the Cura developers. For example, I have only a slight idea what `machine_disallowed_areas` is, or what units it's in, or what changing it might do.

Even if I could figure out how to make this work, Cura should both have a generic delta option added by default, and the ability to edit machine settings through the GUI.

quillford, 2015-07-02:
I would imagine disallowed areas are for things such as binder clips holding glass to a heatbed pcb. As for a centered build area, have the center of the STL be the center of your bed. Pretty simple in OpenSCAD - cylinder(d=<bed_diam>, h=1,center=true);. It is a pretty safe assumption that the units are millimeters. I tried the json and it works but it would be nice if there was an other with GUI adjustable settings.

nallath, 2015-07-03:
Confession here; We -totally- forgot about weird shaped platforms. Our bad. We're looking into it, but I cant give you a good estimate when we will have this. The crash bugs have a higher priority, so we will fix those first.

The machine_disallowed_areas are areas where the nozzle can't print. In the case of the UM2, its the metal clip area above the glass bed. You could hack those to create a sort of circular bed. Won't be perfect but should work. 

The units are in mm.

BagelOrb, 2015-07-03:
There is a machine setting in the json file called "machine_center_is_zero"; try setting it to true.

Adding a new json file for a new printer should work. It does work on my PC. Don't forget to change the name and id at the top of the file.

BagelOrb, 2015-07-03:
@nallath Should we collect the json files for other printers from users and include them in the distribution?

nallath, 2015-07-03:
if they add them with a pull request we can add them in the next dist.

quillford, 2015-07-03:
Instead of having every printer including in the standard distribution, you could just have a Cartesian and Deltabot with adjustable settings and an easy way to load a json profile for a specific printer.

Morgul, 2015-07-03:
@quillford Mostly, I was using that as an example, attempting to illustrate that editing the config files is a rather poor user experience.

@nallath I understand that they got overlooked. :) What gets me, though, is that such a major change was done with absolutely broke backwards compatibility (and in truth compatibility with non-Ultimaker printers), and the version number is still 15.X. Any reason this new version isn't 16.x? Maybe I'm weird, but the rough edges of the current release would be easier to swallow if it was no longer part of the 15.X line.

quillford, 2015-07-03:
@Morgul 15.6 because it was the sixth month (June) and 2015 when it was released.

Morgul, 2015-07-03:
In any event, I think there's three separate, but related issues here (and maybe they should be separate issues in GitHub?)

1) No printers but Ultimaker Printers are available by default.
2) There is no GUI to edit printer properties, such as build volume, center as 0,0,0, etc.
3) Circular build plates are currently not supported.

It seems like `1` just needs someone to send a PR with a "Generic Cartesian" and "Generic Delta". The rest will probably involve a little bit more code.

Morgul, 2015-07-03:
@quillford Huh, ok. I wasn't aware that that was how the versioning worked. (Sorry, I've gotten super used to semver, I didn't realize this was ubuntu style.)

quillford, 2015-07-03:
@Morgul 2 is a big issue IMO. "Generic Cartesian" or Deltabot would be good enough for now.

quillford, 2015-07-03:
Although a JSON/profile loader would probably be easier to manage than having every printer that someone submits a PR for be included. The distribution could include common printers like the Prusa i3 (#97) and Kossel for example.

Morgul, 2015-07-03:
@quillford As long as I can edit the build volume and other parameters for the pinter in the GUI, I think that the list could be as simple as:

* Ultimaker Original
* Ultimaker 2 Go
* ... (All the other Ultimaker printers)...
* Custom

Then, when Custom is selected, we pop up some additional controls to set center as 0,0,0, rectangular/circular build plate, build volume, etc.

Then as along as you can export/share your custom printer settings, we can leave it up to the community to share those settings. At a later date, a system for browsing/adding community defined profiles _could_ be added, if there seems to be demand.

quillford, 2015-07-03:
@Morgul perhaps community defined profiles and plugins can find themselves in some kind of repository accessible from the GUI

Morgul, 2015-07-03:
@quillford That would be an excellent way to handle it.

BagelOrb, 2015-07-06:
@Morgul 
1) We are waiting for pull requests with json files for other printers.
2) A GUI to edit the json file is on our issue list, but has very low priority.
3) Circular buildplates can be implemented with disallowed areas.

nallath, 2015-07-06:
It's very hard for us to tell our managers that we need to spend time making generic printers work when there are tons of bugs that prevent users with an UM to correctly work with Cura. Selling those machines does pay our salary ;) 
Anyone is free to create a plugin that makes modifying the profiles easier of course. If it works well, we might add it to the next release.

Morgul, 2015-07-06:
@nallath I can understand that. It's just very important to remember that Cura has a community beyond UM customers, and some of us were a bit sideswiped by the sudden "support regression", as it were. :)

I'm totally willing to tackle some of this, but I have some questions. First, you see printer options as a plugin, _not_ a modification of existing code? I'm a bit confused as to how that's supposed to work; can plugins add widgets to existing settings pages? Second, The [documentation on plugins](http://ultimaker.github.io/Uranium/master/md_docs_plugins.html) is a little sparse, is that all there is, or could you point me in the right direction (even if it's reading code).

Finally, as I would really like to make sure this makes it in, do the devs have any opinions/thoughts on how they would like this to work? At least as an initial proof of concept, I think that adding a 'machine configuration' section to the Machine settings page seems the most appropriate. After that appears to be working, we can discuss how we should present it in the Add Printer wizard, I imagine.

awhiemstra, 2015-07-06:
Actually, for the moment, printer options are additional resource files. There is currently no way to provide them as plugin. However, as they are individual resource files, there are just additions, no changes to existing code needed. 

For an actual custom machine option, that will require code changes. Currently there is no way to add pages to the "Add Machine" wizard. These are planned to be added for the next release. This will also need tweaks to how the machines are provided as we want to add some of the UM Original configuration pages back, which means they are only relevant for those machines. This might lead to it being possible to add printers through plugins.

quillford, 2015-07-06:
If you guys plan on having a plugin repository managed by Ultimaker, perhaps you can integrate user made profiles in the repository.

awhiemstra, 2015-07-06:
@quillford Something along those lines is what I would like to have, yes. But things like that take time, unfortunately.

quillford, 2015-07-06:
If a GUI feature to edit profiles if of low priority, something like #112 should be in the readme.

nallath, 2015-07-06:
@Morgul; I understand. But all the 'non-um stuff' in the code have always been made by comunity members. This is no different. I do agree that we should have these things. Don't fear, we will help you guys do this, but we're just more open about what we will (and will not) do.

lerouxb, 2015-07-08:
Guys I don't mean to be rude, but this really should have been named "Cura 2 Alpha-1" or something like that with a warning that it is still for power users too.

I just saw a new version of Cura and upgraded like I have many times before, realised that my printer isn't supported, downgraded, lost all my fine-tuned settings again...

And from what I can tell by looking at the forums there are serious regressions and bugs for Ultimaker users too.

People see 15.04 -> 15.06 and just download it and install it, thinking it might fix some bug they are experiencing or maybe add a minor feature or two, definitely not expecting things to break badly.

If it was cura2.app you could at least run it next to cura.app while trying out the latest and greatest. I can tell that this new version is clearly going to be better and it is the future, but it feels like it was released too early and communicated poorly.

I'm a Printrbot Simple Metal user and not a client of Ultimaker and I'm well aware that I'm getting this amazing app for free, so please don't get me wrong. I am REALLY grateful for all the work you guys do. This is really meant as constructive criticism.

nallath, 2015-07-08:
Yeah, I agree. I thought that we were going to release it as open beta. There has been miscommunication on what / how we were going to do this. 

This might sound like an excuse (and people have heard this over and over again) but things like this happen in a growing company. Yes it sucks. Yes we should have known better. Yes we will do it differently next time. We (the software devs) also want something that all of us can be proud of.

closed 2015-07-08 by nallath

Morgul, 2015-07-08:
@nallath For what it's worth, I think the new version is a huge step forward... just a little under-baked at the moment. You guys _should_ be proud.

It can be hard, with an open project like this, to manage user expectations, and yeah, mistakes will be made at a growing company like Ultimaker. I get it, and I completely understand. But I felt it was important to at least let all the devs know that, yes, we still love the software, and we appreciate your work.

That being said, I'm going to dig into this a little this weekend, to see if I can't contribute some of what I want (ui for configuring settings, mostly). I'm still very open to suggestions for where I should look, or how the devs might want to see this done.

lerouxb, 2015-07-08:
I think this just needs a little form that kicks out the json file. Could be on the web somewhere even. Maybe I can make the time for that soon as I really want to play with the new Cura. I think it is totally fine if Cura doesn't have a UI for setting up your own printer's settings built in, btw. Seems totally fair to me.

quillford, 2015-07-08:
@lerouxb I've already written a web one. just haven't committed it to GitHub.

lerouxb, 2015-07-08:
@quillford Cool! Saves me the trouble. Please post a link to it. ;)

quillford, 2015-07-08:
@lerouxb sure :) Just give me a few hours to get back to the computer it is on. Once I do, I'll commit it into [this repo](https://github.com/quillford/CuraProfileMaker).

nallath, 2015-07-09:
@quillford Awesome! You could try and add it to the github pages. That way it's hosted aswell (and everyone can contribute to it).

@Morgul We have making the settings panel dual row /draggable (if width is high enough it shows to collums of settings, bit like old cura) on the road map, so that would be a good place to look. See https://github.com/Ultimaker/Cura/issues/119

quillford, 2015-07-09:
@nallath github pages was the intent :)

quillford, 2015-07-09:
@nallath @lerouxb here it is in a basic form http://quillford.github.io/CuraProfileMaker/. will add start and end gcode among other things a bit later - currently on an iPad.

gearsawe, 2015-08-07:
So after about an hour of messing around, I was able to setup the odd shaped TIKO print volume 
[image] by use of disallow areas. Had to use a CAD program to figure out the cords.

Morgul, 2015-08-08:
@gearsawe: Could you give a brief description of how you did it? It could be very helpful to others, and might even be able to be integrated into @quillford's profile generator. (I've been sending him some patches, and I'm a bit handy with SVG; might code up a disallowed area tool that way.)

gearsawe, 2015-08-08:
Using some cad software I drew the shape of the print area marked its center. Then drew a box with its center the same as the other previous shape. The arcs where converted into straight lines of equal length to approximate the arcs. More lines would mean more accurate. From there just broke it up into quads and started entering the four points of each quad. Note: center is 0,0 even if set the printer has BackRight as 0,0 these point are independed of the printer 0,0.
[image]
[image]
end up with a list of points.
        [[[-92.55,-92.55],[0.00,-92.55],[-16.22,-88.21],[-92.55,-88.21]],
        [[-92.55,-88.21],[-16.22,-88.21],[-28.10,-76.33],[-92.55,-76.33]],
        [[-92.55,-76.33],[-28.10,-76.33],[-80.15,13.83],[-92.55,13.83]],
        [[-92.55,13.83],[-80.15,13.83],[-84.50,30.05],[-92.55,30.05]],
        [[-92.55,30.05],[-84.50,30.05],[-80.15,46.28],[-92.55,46.28]],
        [[-92.55,46.28],[-80.15,46.28],[-68.28,58.15],[-92.55,58.15]],
        [[-92.55,58.15],[-68.28,58.15],[-52.05,62.50],[-92.55,62.50]],
        [[-92.55,62.50],[92.55,62.50],[92.55,92.55],[-92.55,92.55]],
        [[92.55,-92.55],[0.00,-92.55],[16.22,-88.21],[92.55,-88.21]],
        [[92.55,-88.21],[16.22,-88.21],[28.10,-76.33],[92.55,-76.33]],
        [[92.55,-76.33],[28.10,-76.33],[80.15,13.83],[92.55,13.83]],
        [[92.55,13.83],[80.15,13.83],[84.50,30.05],[92.55,30.05]],
        [[92.55,30.05],[84.50,30.05],[80.15,46.28],[92.55,46.28]],
        [[92.55,46.28],[80.15,46.28],[68.28,58.15],[92.55,58.15]],
        [[92.55,58.15],[68.28,58.15],[52.05,62.50],[92.55,62.50]]]

nallath, 2015-08-10:
Wow. I'm impressed!