Ultimaker/Cura #91: Skirt Height

Morgul, 2015-07-02:
This might be more a Cura Engine issue, but it would be nice if Cura supported printing multiple layers of a skirt. This is useful for some warping issues with materials like HIPS, and is a feature in both Slic3r and Simplify3D.

BagelOrb, 2015-07-03:
Yay a CuraEngine issue! :)

Shouldn't be too difficult, but I don't see why/how that should work.
Perhaps you mean multiple layers of brim?
Wouldn't it be better to use a raft then?

Morgul, 2015-07-03:
No, no, I mean skirt. :)

A frequent problem I have with rafts is that they leave the bottom surface rather rough. For some of the parts I print that's less than desirable. With HIPS (due to how it warps) the alternative is to print a sacrificial skirt, like in this image:

[image]

It works really well, but must be multiple layers. It's one of the features that the other big slicers have and Cura doesn't... and you're right, it shouldn't be too hard to add. The uses may be small, but I assure you, they are there. :)

CCS86, 2015-07-04:
Skirt doesn't touch the part. How would that help you? 

Brim is like a skirt, but it connected to the part.

REulberg, 2015-07-04:
I could use this as well.
It acts kinda like a heated build chamber, particularly good at blocking errant breezes generated by heat break cooling and printing motions.

This is very helpful for tall prints, with think walls, printed wirh high warp materials.

CCS86, 2015-07-04:
Ahh, I get it now

BagelOrb, 2015-07-06:
Indeed, thanks for explaining.

I suppose it doesn't need to be the skirt exactly. I already know some nice optimizations making such a heat chamber more nice :)

Currently working on some more pressing issues, though. I hope I can get the feature implemented in a month or so.

BagelOrb, 2015-07-08:
Do you know what the feature is called in slic3r and s3d?

BagelOrb, 2015-07-08:
In slic3r and s3d it's called "skirt height", which I think is a misnomer given the functionality and the way it is computed.

I think I will call it "draft protection".

Morgul, 2015-07-08:
@BagelOrb I'm fine with that, though "skirt height" is literally what it is: "Draw up to X layers of skirt".

Question, will the skirt always be model height, or will the height be configurable? If it is configurable, please make sure it's 'up to X layers', and cuts off at the model. :)

BagelOrb, 2015-07-08:
It is not just the skirt, because the skirt is just around the first layer, while the draft shield has to avoid other layers as well.

The height can be set in millimeters and stops at the last layer of the model.

[ IMPLEMENTED ]
:D

closed 2015-07-08 by BagelOrb

Morgul, 2015-07-08:
@BagelOrb I just meant in slic3r's case. However, this is exciting!

ChrisKrieg, 2015-10-12:
Sorry if this is a stupid question. Can I access this new draft shield setting in CuraEngine via Repetier-Host? There is no skirt height setting in the most recent Repetier Cura tab.

Thanks.

Ghostkeeper, 2015-10-13:
Repetier-Host uses an old version of CuraEngine, since otherwise Ultimaker might break Repetier's software from time to time with updates. Since then we've changed a few things, including a lot of setting names, that would break Repetier-Host.

ChrisKrieg, 2015-10-14:
So how do I go about using the version of Cura that contains the draft guard option?  Is there a development .exe or installer that I can try on my machine?

Sorry if I'm wasting your time. I'm a noob to git.

BagelOrb, 2015-10-15:
You can download an old beta of the new Cura at https://ultimaker.com/en/cura-software/list.
Otherwise search the forum for the latest beta.

ChrisKrieg, 2015-10-15:
Thanks a lot! I've tried 15.06, 15.04, and beta builds 15.09.084, 15.09.85, and 15.09.90. None of those versions generates the draft shield when it is selected. All other settings seem to work, except for Raft, which does not generate.

Ghostkeeper, 2015-10-16:
We noticed that about the raft, and it has been fixed just this week :smiley:

BagelOrb, 2015-10-16:
I think the problem is that the draft shield just isn't visualized.

BagelOrb, 2015-10-16:
There seem to be several problems:
- the draft shield is not visualized
- the draft shield height isn't set to 99999 when Limitation is set to Full
- the raft margin is added to the boundary of the draft shield; it should be the ground parts offsetted with the raft margin unioned with the draft shield (and other helper parts) polygons offsetted by 1.0mm hardcoded.

ChrisKrieg, 2015-10-16:
The draft shield is not only unvisualized, it is not generated. I exported the gcode and opened it in Repetier to preview and the guard paths aren't present.

Just reporting what I see. Thanks for your hard work!

BagelOrb, 2015-10-16:
The second problem I noted covers your problem.