Ultimaker/Cura #100: Model jumps when dragging to new position

Mitrasmit, 2015-07-03:
When dragging a model across the platform, the view sometimes jumps to a different position, taking the model with it (sometimes even forcing the model off-screen). 

It seems to happen when you deselect the model, then select it and immediately drag it. It wants to reposition the camera view with the product centered, but when you are holding the mouse button, the whole position of the model on the platform changes.

irrenhaus, 2015-07-06:
+1 here

TMHogenhout, 2015-07-20:
<h3>User feedback: </h3>
It seems that the model jumps to a position relative to the pointer where it should be relative to the build-plate centre. When you load two models, the model the farest from the centre, gets the biggest displacement in the opposite direction of the build-plate centre. De model goes back to the right position after the user has clicked one of the arrows, after that the displacement starts building up again.

nallath, 2015-07-21:
This was caused by another fix, we're working on fixing it.

fieldOfView, 2015-09-01:
This seems to fix it for me: https://github.com/Ultimaker/Uranium/pull/65

closed 2015-09-02 by nallath