Ultimaker/Cura #101: Print One At a Time - functionality disappeared?

Mitrasmit, 2015-07-03:
Is it true that the option to print several models after each other, instead of at the same time, has been removed from the newest version? Or can I just not find it?

daid, 2015-07-06:
It's one of the features which has not been implemented in the new GUI yet, but is planned for the future release.

Until then, if you need this feature, you'll have to use the previous release:
http://software.ultimaker.com/current/

closed 2015-07-06 by nallath

BagelOrb, 2015-07-06:
Does that mean it is now implemented in the GUI already, or that it's not going to be done?

nallath, 2015-07-06:
Its not implemented, which we communicated. It will be implemented in the future.

daid, 2015-07-06:
(closed because it's already on the internal feature development list for future planned features, also multi material printing will depend on the details of the implementation of this feature, so that's why it will be done for sure)