Ultimaker/Cura #102: Cura did not load files provided by Fusion 360

Spirou42, 2015-07-03:
Steps to Reproduce:
1. have a Mac with Fusion360 installed (test version will do)
2. Create an object 
3. Try to 3d Print the object 
3.1 Menu: File->3d Printing
3.2 Select Object
3.3 Set  Print utility to the Cura 15.06.1 and hit return:

Result: Cura starts 
Expected Result: Cura starts and loads the STL.

This is a regression to Cura 15.02.1

awhiemstra, 2015-07-06:
I expect this is a difference in handling command line arguments that are passed along on the command line. As a workaround, can you try calling the application directly from Fusion360, it is in Cura.app/Contents/MacOS/Cura .

nallath, 2015-07-08:
@Spirou42 Did the work around posted by awhiemstra work?

Spirou42, 2015-07-08:
@awhiemstra: indeed, this is most likely a command line argument parsing issue.
Unfortunately there seams to be no way i'm aware of to trick fusion360 to call the executable from the applications bundle directly. 

1. The default "file selection dialog" (popping up if I select the "Custom" entry in Fusions 3d printing dialog)  prevents me from navigating into the applications bundle. 
2. trying to drop the Cura executable (from inside the applications bundle) to the "file selection dialog" at least bring me to the right directory BUT the executable is not selectable.

3. The standard preferences ~/Library/Preferences/com.autodesk.fusion360 only contains stuff from the  QTLibrary and "normal" NS* entries from the WindowServer.  
4. Searching for the "real" preferences inside the Funsin360 installation directory (below ~/Library/ApplicationSupport) has failed till now I have not found the "real" preferences file.

@nallath: 
1. see above
2. i would not classify this as an enhancement but as a regression :)

nallath, 2015-07-08:
This is an enhancement for this version of Cura, especially as 15.04 is completely different from 15.06 (it's a complete rework of the 'old' cura).

Spirou42, 2015-07-08:
@nallath Thats a pure technical argumentation and does not take the Users point of view into account. From my point of view it does not matter if you have rewritten the whole application or parts or what ever. This is a pure technical decision and as a software engineer i can even understand your argumentation BUT it still is pure technical.
As a user i'm not interested in the technical details of the softwares implementation, it merely is completely irrelevant what you have done to the implementation. For me as a user of a Software which did not changed its name and did not changed version numbering etc. only existing features or the lack of them matter. 
Therefor, from my point of view (as a user), each and every feature missing in version 15.06.X is a regression.  

If you take your argument one step further, there can't be any regressions at all. Cause if you have rewritten a features implementation it is new and therefore no regression can exist. 

On the other hand it is Ultimakers decision how you'll call now brocken or missing features but you have to be aware, that Users might completely disagree with your point of view.

nallath, 2015-07-08:
The labels are just for us to see what is going on. The github issue tracker system sucks, so we can't label things in exactly the way we want it. 

Fusion 360 support wasn't even an official feature (I didn't even know that it was possible in the first place) in the first place that I'm aware of. So you're free to call it not working whatever you want (regression, bug, damn you developers this should work, chair, etc) ;) 

Also; This is still an open source project. Ultimaker only decides what the resources they put on this project (Me and awhiemstra) get to work on. Others are free to also put resources in it and build whatever the hell they want.

Spirou42, 2015-07-08:
@nallath You two have done a nice job on Cura  - so don't get me wrong i don't want to call you rude names and i totally agree with you, that the github issue tracker is a "nice try". 

Back to the original issue:
If I start Cura from the command line either with _open \<CuraApp\> --args \<STLFILE\>_ or using the executable directly with out _open_ and _--args_ both versions  of Cura behave the same. There has to be another reason why the versions are not compatible in this situation. 
I'll try to get more information on how Fusion tries to call its external support applications.

nallath, 2015-08-14:
Just a quick update; One of the developers of fusion360 posted on our forum to ask for contact details. We're trying to get into contact with them.

TMHogenhout, 2015-11-13:
Fixed for the next release

closed 2015-11-13 by TMHogenhout