Ultimaker/Cura #103: Curt can't load AMF files created by legacy Cura

Spirou42, 2015-07-03:
I have a bunch of STL/AMF files created by Cura 15.02.1. 
Cura 15.06 can read any of these files any more.

Steps to reproduce:
1. save a STL/AMF in cura 15.02.1 
2. start Cura 15.06 and try to load the files
Result: endless dialog (at the bottom of the cure window) telling me, that cure tries to read the files
Expected: load the files or at least present a meaningful message why things screwed up.

awhiemstra, 2015-07-06:
AMF files are expected not to load, we currently do not have an AMF reader plugin. If the STL files also do not load correctly, please share the model it refuses to load.

Spirou42, 2015-07-06:
Exporting the AMF to STL with Cura 15.02.01 get the files readable again - 
So, not reading AMF is a regression and not informing the user about a unrecognised/unsupported format and endless displaying of the load progress window is a bug.

nallath, 2015-07-06:
Agreed, it should notify the user of unsupported format.

nallath, 2015-08-24:
Failing to load is now correctly handled. As long as there is no official AMF document (eg; They stop being so -damn- secretive about what they are doing) we won't re-add AMF support.

closed 2015-08-24 by nallath

awhiemstra, 2015-08-24:
Additionally, we can now load 3MF files. This format will most likely replace AMF in the longer run.

Patola, 2016-01-18:
People, I know this ticket is closed and this regression is most likely not being worked on. But PLEASE, reconsider supporting AMF even if in its most simple form. There are people with needs for dual/triple extrusion like me and the Cura interface is just the best, hands-on. I understand the grude with "no official AMF document" but I use Linux and have no tools to work with 3MF, but I have for AMF. And loading two or three STL files sometimes just won't work and be portable enough.

Ghostkeeper, 2016-01-18:
In our internal bug tracker the issue is approved for "some future release". There's no saying what we'll do with our free time though, and this is one of the things I had in mind. And of course, if someone in the community wants to have a go at it that'd be fine too.

ianrrees, 2017-02-07:
I'm curious what is meant by "no official AMF document" - there clearly _is_ an official AMF standard document, ISO 52915:2013, so is the issue that there is nothing freely available online?

nallath, 2017-02-08:
There are a few reasons why we won't do it:
1. There is little request for AMF support.
2. There is a good alternative for AMF (eg; 3mf)
3. Last time we checked, there is no publicly available AMF standard. This is fine, but it will hamper the development of the standard. This point is tied in with point 1.
4. When making the code for AMF in old Cura (back in the day), other devs noticed that there are multiple conflicting statements in the AMF spec. Their own examples for instance are not even valid.

As for the request by @Patola; I've found a plugin that gives blender the ability to import / export 3mf (https://github.com/solbaver/blender_3mf_import_export)

ianrrees, 2017-02-08:
Thanks for the response @nallath - was really just asking about your point 3.  I think my confusion comes from "standard" being used where I would've used "reference implementation", so I got a bit mixed up.

thopiekar, 2017-02-08:
@nallath Why are you pointing @Patola to blender plugin/scripts? Are these blender files these AMF files you are talking about here? I'm a bit confused..

nallath, 2017-02-09:
@thopiekar He was talking about that none of his linux software can handle 3mf, but can handle AMF. I'm not sure what he wants, but blender does run on linux and can handle 3mf.